ArduCAM/ArduCAM_USB_Camera_Shield #5: Unable to Open MT9V034

chengyuen93, 2017-08-22:
Hi, I followed the instruction in README.md and installed libusb and python2.7, but when I try to run the script it keeps telling me that "open device failed". Do you know what could be the issue?

Thanks

ArduCAM, 2017-08-22:
You have to add sudo before the commands.
 
From: chengyuen93
Date: 2017-08-22 12:16
To: ArduCAM/ArduCAM_USB_Camera_Shield
CC: Subscribed
Subject: [ArduCAM/ArduCAM_USB_Camera_Shield] Unable to Open MT9V034 (#5)
Hi, I followed the instruction in README.md and installed libusb and python2.7, but when I try to run the script it keeps telling me that "open device failed". Do you know what could be the issue?
Thanks

chengyuen93, 2017-08-22:
Hi

Yes I put sudo in front of the commands. But it still shows open device failed. Should /dev have any indication that the camera is connected?

ArduCAM, 2017-08-22:
You can use lsusb to check if there is any new USB devices detected.
It should be 52cb:52cb.

chengyuen93, 2017-08-23:
Below is before and after I plugin the shield:

Before:
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 04f2:b374 Chicony Electronics Co., Ltd 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

After:
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 04f2:b374 Chicony Electronics Co., Ltd 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
**Bus 003 Device 002: ID 04b4:8613 Cypress Semiconductor Corp. CY7C68013 EZ-USB FX2 USB 2.0 Development Kit**
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

So it seems like I got this 04b4:8613 Cypress Semiconductor Corp rather than 52cb:52cb for MT9V034.

What could be the issue and is there a way to solve this?

ArduCAM, 2017-08-23:
Thanks for your information.
It seems that the firmware was lost due to some reason.
Would you please let me know your email address, I will send you firmware update guide by email.

chengyuen93, 2017-08-23:
Here is my email:
cedric_ong93@hotmail.com

Thank you

chengyuen93, 2017-08-23:
Hi

I followed your instruction as you mentioned in your email (and with the camera connected). Resintalled the driver (Cypress FX2LP No EEPROM, which is the only option). But when I upload the .icc file you attached in the email, it says programming failed. Is there a way to solve this problem?

Thank you very much for your help.

Regards
Cheng Yuen

chengyuen93, 2017-08-23:
just to confirm, I am supposed to short the jumper before power up, and once the camera is power up I have to remove the jumper. Then proceed on to updating the firmware right?

ArduCAM, 2017-08-23:
Yes, you are right.

chengyuen93, 2017-08-23:
Yes, then is there a possible reason as to why it says programming failed? thanks.

chengyuen93, 2017-08-24:
Hi, I tried reinstalling both win7 and win8 (not sure if it has got anything to do with the driver but i was just trying it out anyway) on my win10 laptop, and I tried quite a few times and it still says programming failed.

[image]

Do you know what could be the reason that it is failing? thanks.

chengyuen93, 2017-08-24:
Hi

Loading .hex to RAM says programming succeeded! However, when I tried to auto-open it using USBDemo.exe, the software still crashes.

[image]

and also if I want to run it using a python script in Linux, how can i do it since I have to update the firmware in Windows.

Thanks.

ArduCAM, 2017-08-24:
Can you see any new device found after loading the firmware?

chengyuen93, 2017-08-25:
Yes, which is Cypress FX2LP No EEPROM Device

closed 2017-08-29 by chengyuen93