ArduCAM/ArduCAM_USB_Camera_Shield #6: ArduCamConfigLib.dll requires mfc100d.dll and msvcr100d.dll

wt-bqng, 2017-09-26:
I would like to request for the ArduCamConfigLib.dll to be compile in Release version instead of Debug version. As the debug version will require the users to have Visual Studio 2010 installed in the PC in order for the Windows/USBTest software to run.

ArduCAM, 2017-09-26:
They are released version actually, you have to install the vcredist_x86.exe or vcredist_x64.exe for the missing lib support.
https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Windows/USBTest/x64/Release 

From: wt-bqng
Date: 2017-09-26 11:15
To: ArduCAM/ArduCAM_USB_Camera_Shield
CC: Subscribed
Subject: [ArduCAM/ArduCAM_USB_Camera_Shield] ArduCamConfigLib.dll requires mfc100d.dll and msvcr100d.dll (#6)
I would like to request for the ArduCamConfigLib.dll to be compile in Release version instead of Debug version. As the debug version will require the users to have Visual Studio 2010 installed in the PC in order for the Windows/USBTest software to run.

wt-bqng, 2017-09-26:
Running DependencyWalker on ArduCamConfigLib.dll shows that mfc100d.dll and msvcr100d.dll are needed. Take note that both end with the "d", which stand for debug version.

Installing the vcredist_x86.exe or vcredist_x64.exe only provides mfc100.dll and msvcr100.dll (without ending with d).

ArduCAM, 2017-09-26:
Thanks for your reminder.
I will fix this issue ASAP, please be patient.

ArduCAM, 2017-09-27:
I have update the library with released version, please try it again.
Thanks for your report.

closed 2017-09-27 by wt-bqng