ArduCAM/ArduCAM_USB_Camera_Shield #10: MT9J001 no longer recognized

andrewalken, 2017-10-02:
The USB test no longer reads the MT9J001, 
When I click auto-open i get the message "USB device create  error" 
And when i Click open I get an error "incorrect Parametes" 
[image]

ArduCAM, 2017-10-03:
You have to scan the device before clicking open, or you can use auto-open instead.

andrewalken, 2017-10-03:
Scan does nothing and like I said, when I click 'auto-open' I get an error saying 'USB device create error'

ArduCAM, 2017-10-04:
Thanks for your info.
It seems that the device driver is not installed properly.
Please check the Windows device manager if you can find any new USB device and check the driver.
www.arducam.com/downloads/shields/USB_Shield/ArduCAM_USB_Camera_User_Guide.pdf
And some notes about the installation drivers on windows: https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/blob/master/Drivers/README.md 
Once you installed the driver, you can evaluate the camera according to this application note: 
www.arducam.com/downloads/modules/industrial/ArduCAM_USB_Camera_MT9J001AN.pdf

andrewalken, 2017-10-04:
Hi Lee

Thanks for that!

You were right about the driver,

Many thanks
Andrew

closed 2017-10-04 by andrewalken

andrewalken, 2017-11-07:
Hi Lee,

As you know I'm working with Bryan Hennelly using the MT9J001 Sensor,
Due to my neglect I have scratched the surface of the sensor and was wondering how much it would be to purchase a new sensor from you.

Regards,
Andrew

On Wed, Oct 4, 2017 at 2:49 PM, Andrew Alken <andrewalken@hotmail.com<mailto:andrewalken@hotmail.com>> wrote:
Hi Lee

Thanks for that!

You were right about the driver,

Many thanks
Andrew

andrewalken, 2017-11-20:
Hi Lee,

Would you be able to send me the SDK for the camera shield with the MTJ9001 sensor please.
I couldn't seem to find it online.

Many Thanks
Andrew

On Tue, Nov 7, 2017 at 12:19 AM, Andrew Alken <andrewalken@hotmail.com<mailto:andrewalken@hotmail.com>> wrote:
Hi Lee,

As you know I'm working with Bryan Hennelly using the MT9J001 Sensor,
Due to my neglect I have scratched the surface of the sensor and was wondering how much it would be to purchase a new sensor from you.

Regards,
Andrew

On Wed, Oct 4, 2017 at 2:49 PM, Andrew Alken <andrewalken@hotmail.com<mailto:andrewalken@hotmail.com>> wrote:
Hi Lee

Thanks for that!

You were right about the driver,

Many thanks
Andrew

ArduCAM, 2017-11-20:
The github already support the MT9J001, please check the configuration file from: https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Windows/USBTest/x64/Release/Config

andrewalken, 2017-11-20:
Hi Lee,

I understand it supports it but I was told there was an sdk in C

And I cannot find it on github

ArduCAM, 2017-11-20:
The SDK is provided in dll and lib format, you can find them from https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Windows/USBTest/x64/Release 
You can use USBTest demo as a start point to develop your software using our cameras.