ArduCAM/ArduCAM_USB_Camera_Shield #9: MT9F002 capture failing on Windows and Raspberry pi

tbttfox, 2017-10-02:
On Windows 10 64:
I installed the driver properly, and opened the camera demo.
Inside the demo I can choose the MT9F002, and `auto-open` the camera successfully.  It reports `USB camera init success!`.  However, when I try to hit the `play` button, all I get is:
```
[22:39:56]USB transfer data length error
[22:39:55]USB transfer data length error
[22:39:55]USB transfer data length error
[22:39:54]USB transfer data length error
[22:39:54]USB transfer data length error
[22:39:54]USB transfer data length error
```
Hitting `shot` with the with the camera open does nothing.
---

On the Raspberry pi:
After getting all of the dependencies detailed in the README, and running the line:
`> sudo python MT9F002.py --type jpg --name sample_image`

This gets printed to the terminal (including the leading space in line 3)
```
device open success!
capture fail!
 is not available
data length is not enough!
is not available
data length is not enough!
device close success!
```

ArduCAM, 2017-10-02:
We just update the windows demo software, please try the new software and try it again.
https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Windows/USBTest 
Let me know if you make any progress.

tbttfox, 2017-10-03:
Ok, lots more options. Good!  Now I'm getting this with both the 8 and 12 bit options:
```
[10:10:56]Bad frame received
[10:10:55]Bad frame received
[10:10:55]Bad frame received
[10:10:54]Bad frame received
[10:10:53]Bad frame received
[10:10:48]USB camera init succes
```
And the "force display" option gives me a slightly noisy black background with a true-black vertical line.  I can send an image later when I have a little more time.

ArduCAM, 2017-10-04:
Can you show me more screenshot of the issue?
You can try to reconnect the camera to see if making any difference.
Here are two more settings for MT9F002, please copy into the Config folder.
[www.arducam.com/downloads/modules/industrial/MT9F002/MT9F002_VGA_8bit.cfg](url)
[www.arducam.com/downloads/modules/industrial/MT9F002/MT9F002_1080p_8bit.cfg](url)

tbttfox, 2017-10-04:
Thank you very much for your help.
Now that I have had more time to check, it seems that the USB port was to blame.  Plugging into the port next to it fixed the issue.

Also, I appreciate the extra configurations. They helped me realize that the display area does not auto-fit to the screen size.  The "Full Screen" checkbox should probably be called "Fit to Window", or something similar.

However, the only reason I was using Windows was to make sure the board worked. Getting it working on the Pi is my ultimate goal.  Now that I have verified that my camera is working, that is where I will be spending my time.

ArduCAM, 2017-10-04:
I'm glad to hear that you have fixed your problem. We are also working hard on improving the performance on Raspberry pi board like the up-to-date Windows version. We will update the RPI version very soon. Let me know if you need any help from us.

closed 2017-12-15 by tbttfox