ArduCAM/ArduCAM_USB_Camera_Shield #11: Device open fail

Moinelly, 2017-11-28:
I have followed the instruction's on Read.md and followed the  steps in linux but when i run the script MT9V034.py , i get a message like -"Device open fail!"
Is there anyway to solve the issue.
Thank you!

ArduCAM, 2017-11-28:
Please try the windows demo software first to make sure hardware setup is correct before moving to Raspberry pi.
Let me know if you still have problem.

Moinelly, 2017-11-28:
when trying with USB demo test in windows,
I get

> [16:08:57]USB device create error
[16:08:49]USB device create error
following message

ArduCAM, 2017-11-28:
Make sure if you install the device driver correctly.
Please check the windows device manager if the device is correctly installed the driver.

Moinelly, 2017-11-29:
Camera is working with Windows USBtest.exe but failed to run in raspberry pi

> "data length is not enough!
is not availiable
data length is not enough!
is not availiable
device close success!
data length is not enough!
Segmentation fault (core dumped)
!"
This is the error i'm getting while running in raspberry pi

>

Moinelly, 2017-11-30:
Thank you for your support,I can get it working now.
I have gone through the python code can you explain me how can i use AR0134 in trigger mode and what changes i have to make in the code.

ArduCAM, 2017-12-01:
The TRIGGER mode of operation requires that register R0x301A (Reset Register) bit 2 be
set to “0”. Setting this register to “1” will switch the sensor back to the master mode of
operation. The general purpose I/O (GPI) pins must also be enabled

R0x301A Reset Register Bit[2] Stream 0 = Trigger mode
R0x301A Reset Register Bit[8] GPI_EN 1 =GPI input buffers enabled
R0x301A Reset Register Bit[11] forced_pll_on 1 =PLL always powered

Moinelly, 2017-12-01:
I have changed the bits of R0x301A as you said, can i trigger the camera using hardware or is it a software trigger

ArduCAM, 2017-12-01:
You have to trigger the camera with the hardware pin marked "Trigger" on the camera board.

Moinelly, 2017-12-05:
I have tried giving input to 0x301A as 0x1989 but i'm getting Error as Bad frame received can you tell me exactly what will be the input to 0x301A in terms of HEX value for Trigger mode of camera

ArduCAM, 2017-12-05:
we set the 0x301A with value 0x1982 to enable external trigger mode, when enter the external trigger mode, there is no frame coming into the USB camera shield unless you feed the hardware pluse in the trigger pin. The reason why the software report "UNKNOW ERROR", just because it can't receive any data during a period of time, the software is confused cause it doesn't know the camera working in external trigger mode.
All you need to do is feeding the pluse in the Trigger pin on the camera board.
Hope this help.

Moinelly, 2017-12-05:
I have done as per your saying and i have changed the flash register also 0x3046 - 0x01C0...
but if i left the trigger pin open (without connecting) it still getting frames but at a lower rate and what would be the minimum pulse duration for each frame

Moinelly, 2017-12-07:
Hey Lee,
Everything is working good in Trigger mode but the problem is i have three camera Modules of AR0134,out of which one is working good and the rest two have problems.
One camera module gives out -UNKOWN ERROR every time i put in Trigger mode and when i check the Data in 0x301A it gives out 0x1980 where as i put 0x1982.
and the other camera module don't display anything it gives 0X58D every time and on every Address.
Is there is any solution to work with this Two camera Modules.
Thank you!

ArduCAM, 2017-12-07:
1. Please make sure all the cameras working on auto trigger mode, before moving to external trigger mode.
2. "the other camera module don't display anything it gives 0X58D every time and on every Address.", please resit the FPC cable and try it again, it might be loose.

Moinelly, 2017-12-07:
Regarding the Point 1 you mentioned I have checked in Auto trigger mode it is working fine and the problem i encountered is the camera worked in Trigger mode and Auto trigger mode but the flash is not working in both modes.
And point 2 ,i have reset the FPC cable and tried in every possible way but it is of no use.
Thank you for responding fast..

ArduCAM, 2017-12-07:
1. The AR0134 provides an output pin, FLASH, to indicate when the exposure takes place.
When R0x3046[8]is set, FLASH is HIGH during exposure. By using R0x3046[7], the
polarity of the FLASH pin can be inverted
2. 1 out of 3 cameras is not working? Please contact our sales for replacement and send back the faulty camera.

Moinelly, 2017-12-07:
Regarding point 1 i have done that first i have set 0x3046 to 0x0100 still i didn't get any flash and also changed it to 0x01C0 still no use.
and regarding 2nd camera flash is working but it doesn't work in Trigger mode.
I have 4 cameras of AR0134 out of which 1 is working fine and 1 is damaged as you said and the rest two cameras i have found this faults like for one camera Trigger mode is not working and for the other Flash is not working. (Strange)
Point-2. i will contact you sales department.

Moinelly, 2017-12-07:
hey,
Flash is working with the camera,i used 0x0100 but the major problem is when i change the mode from auto to Trigger it gives - "UNKNOWN ERROR" and when i check the data in 0x301A i get 0x1980 where as i enteres 0x1982.
What could be the problem in Trigger Mode.

ArduCAM, 2017-12-07:
As I mentioned eariler "UNKNOWN ERROR" is correct when you switch to external trigger mode, all your need to feed the pulse signal to the trigger pin on the camera board.

Moinelly, 2017-12-07:
Ok i will try that , But what about the other camera it is working in External Trigger mode but it doesn't flash.

ArduCAM, 2017-12-07:
Please make one working first before fixing the others.

Moinelly, 2017-12-07:
ok,Thank you.
I will try with one camera

Moinelly, 2017-12-07:
I am trying the camera in trigger mode can you tell me why there is a change in data when i entered 0x1982 in 0x301A it is showing me 0x1980 similarly when i entered 0x01C0 in 0x3046 it is showing 0x8180.Why is the data changing.

ArduCAM, 2017-12-07:
You have to read the document: www.arducam.com/downloads/modules/industrial/AR0134_RR_D.PDF 
0x301A bit[1]: restart
This bit always reads as 0. Setting this bit has two effects: first,
the current frame is read out and the sensor enters standby.
Second, any writes to frame-synchronized registers and the
shutter width registers take effect immediately, and a new
frame starts. The current frame completes before the new
frame is started, so the time between issuing the Restart and
the beginning of the next frame is a maximum of tFRAME

0x3046 bit[15]: read only, bit[6:0]: Reserved,

Moinelly, 2017-12-07:
I have gone through the datasheet which you have sent and tried putting 0x0100 in 0x3046 to enable flash, and connected a led to the flash pin wrt to ground and attached a multimeter but all i get is 0 volt and led is not flashing.
I even checked whether data and it shows me 0x100. but led is not flashing and i can't get any readings on multimeter.
Thank you!

Moinelly, 2017-12-07:
I am attaching a image and screenshot of what i have done.
[image]
I have connected Flash pin and ground pin to led's.
[image]
I have added 0x0100 in 0x3046 to enable Flash everything seems to be correct but the led's are not flashing.
What is the Problem i'm facing right now to solve this.

ArduCAM, 2017-12-07:
Thanks for your info.
I will work on this during the weekend.

Moinelly, 2017-12-07:
This is the problem with the 2nd camera which i am facing now
[image]
I have given the  pulse to External Trigger still i'm getting -"UNKNOWN ERROR".
and i have even checked the datasheet also is there any limit in voltage for giving trigger or any other factors to make it work.

Moinelly, 2017-12-08:
Hey Lee,
I have a small doubt regarding trigger mode,what is the minimum and maximum voltage requirement to give a trigger to trigger input.
What i think is the voltage to trigger pin is not sufficient that's why there is an -UNKNOWN ERROR.
What do you think the problem can be,because i need a perfect working camera cuz i have a deadline to complete it.
it would be very helpful if you could help me on this issue to resolve it .
Thank you!

ArduCAM, 2017-12-08:
The trigger input voltage is 3.3V TTL.
I have been messed up by your different behavior of different camera modules.
Please send me email to clarify: admin@arducam.com

closed 2017-12-25 by Moinelly

coldstart01, 2018-01-25:
Hi Lee, first time commenting here. I have the Arducam MT9V034_VGA & USB sheild. Please help! Somewhere in the 2 days that I have had this the image became inverted. I don't know how to reset the sensor. Also the FPS never changes: I can set register 11 to 3 or 0x20 or 0x40 and its always reading back a framerate of 37 - advertised of course at 60.

Moinelly, 2018-01-26:
@coldstart01 can you give more information,Are you using that camera in Windows right?

ArduCAM, 2018-01-26:
See this thread: https://github.com/ArduCAM/Arduino/issues/292#issuecomment-360662384

coldstart01, 2018-01-26:
> @coldstart01[1] can you give more information,Are you using that

Yes, windows x64. USB 2/3

Links:

  1. https://github.com/coldstart01
  2. https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/issues/11#issuecomment-360714732
  3. https://github.com/