ArduCAM/ArduCAM_USB_Camera_Shield #12: Regarding MT9V034

Moinelly, 2017-12-11:
Hey Lee,
I have started to work on MT9V034 and so far it is going well in window's. I can control the camera in Trigger mode and auto-trigger mode,can change the exposure timing etc..
but the problem is everything is working good in windows but when i run the python script in Raspberry pi
i can't see anything all i can 

> "device open success!
transfer task create success!
capture fail!
device close success!"
but when i use AR0134 i am getting a display with 3 to 4 fps but i can't get anything in MT9V034

ArduCAM, 2017-12-12:
Try to add this line to the example code: 
ArduCam_writeReg_8_8( useHandle, 0x46, 1, 0x05);

Moinelly, 2017-12-12:
i have tried to insert that line in example code,i'm getting an error

>   File "MT9V034.py", line 354, in <module>
    video()
  File "MT9V034.py", line 329, in video
    ArduCam_writeReg_8_8( useHandle, 0x46, 1, 0x05);
NameError: global name 'ArduCam_writeReg_8_8' is not defined"

Where to add this line in code

Moinelly, 2017-12-12:
Thank you!
It worked `ArducamSDK.Py_ArduCam_writeReg_8_8( handle, 0x46, 1, 0x05);`,I used this

Moinelly, 2017-12-12:
Hey lee,
I have used that line it worked for 2-3 times and again i can't see anything the same error is repeating.
can you tell me where exactly i have to add the following line to make it work everytime.
Thank you!

Moinelly, 2017-12-23:
I have tried the demo_sdk which you have sent to me and i followed your instruction's on MT9V034_demo.cpp
the error i'm getting is :

> /usr/bin/ld: skipping incompatible ./libArduCamLib.so when searching for -lArduCamLib
/usr/bin/ld: skipping incompatible ./libArduCamLib.a when searching for -lArduCamLib
/usr/bin/ld: cannot find -lArduCamLib
collect2: error: ld returned 1 exit status

what will be the solution for this,i have tried it  in linux and raspberry.
In raspberry pi,there is no error but it i can't see any messages in Terminal.

Moinelly, 2017-12-24:
I have used your MT9V034_demo.cpp code in raspberry_pi. It is compiling error free but when i run the Test file. i get error

> ./Test: error while loading shared libraries: libArduCamLib.so: cannot open shared object file: No such file or directory

I have tried - **ldd Test**,and in the response i got 

> libArduCamLib.so-not found

and i have tried to update the cache by "sudo ldconfig -v"
nothing is solving problem,What could be the solution,can you help me in this how to locate/add shared library
ASAP

ArduCAM, 2017-12-24:
Please copy these .so files to your /lib and /usr/lib folder.

Moinelly, 2017-12-24:
It worked ./Test is working but the error is,

> *** Error in `./Test`: double free or corruption (fasttop): 0x00410ae8
Aborted

what is this error,I have not made any changes in the code and i copied all the .so files in /lib and /usr/lib folder

ArduCAM, 2017-12-24:
run the demo with command "sudo MALLOC_CHECK_=0 ./Test"

Moinelly, 2017-12-24:
Now the code is working and getting struck, It is not going inside while loop.

> Device found = 1
ArduCam_open successful

after this it get struck's and nothing is displayed, i checked the value of handle = 0x6f800878

Moinelly, 2017-12-24:
sudo MALLOC_CHECK_=0 ./Test worked perfectly,The problem now arises is after "ArduCam_open successful" nothing is displayed
i.e the code is not going in while loop and stuck there.
what is problem here?and can you please provide the solution.
Thank you!

Moinelly, 2017-12-24:
[image]

I'm stuck here!can you please provide the solution because the same thing is happening when i use demo python code for MT9v034, once in a blue moon it's displays the image after that it get stuck in the middle.
Please provide the solution to this problem.
ASAP
Thank you very much

ArduCAM, 2017-12-24:
Make sure if the hardware is well connected in Windows before moving to RPI.
It seems that your camera is not well connected.

Moinelly, 2017-12-24:
I have checked it in windows. It is working perfectly but when i shift it to RPI the problem is encountered.
i didn't change the hardware or software.
AR0134 is working in RPI but the problem is MT9V034. like i said once in a blue moon it works perfectly in RPI later when i test it. i encounter the same problem.
What is the solution?
Thank you!

ArduCAM, 2017-12-24:
What is blue moon?
It is 23:00 here, too late. I will check it tomorrow.

Moinelly, 2017-12-24:
It just a phrase,what i was trying to say is the device works randomly,and when i check it again it stuck.
we will solve it tomorrow.
Thank you for the help!

Moinelly, 2017-12-25:
Hey Lee,
I have checked MT9V034 in Windows it is working perfectly but when i shift the hardware to Linux/RPI, I'm getting the same problem.
The code is stuck and nothing is working.I have checked it in python code and cpp code same problem but i didn't get any problem when i tested AR0134.
as far as my knowledge the code is going in While loop(python code) after that it get stuck and even i don't get any response from the thread.
What could be the problem?
Can you provide a solution ASAP.
Thank you

ArduCAM, 2017-12-25:
Please try the update example for MT9V034
www.arducam.com/downloads/app/ArduCAM_USB/RPI/MT9V034_RPI_SDK_Demo.tar.gz

Moinelly, 2017-12-25:
Thank you!
It's working now,can you tell me what was the problem occurred here?

Moinelly, 2017-12-26:
i have made following changes in python code to make it work in trigger mode,the problem is the camera is going in trigger mode but not displaying anything.
`def video():

	global flag,regArr,handle

	data = {}

	regNum = 0

	res,handle = ArducamSDK.Py_ArduCam_autoopen(cfg)

	ArducamSDK.Py_ArduCam_writeReg_8_8(handle, 0x46, 1, 0x05)

	ArducamSDK.Py_ArduCam_writeSensorReg(handle, 0x03, 0x1E0)

	ArducamSDK.Py_ArduCam_writeSensorReg(handle, 0x04, 0x2F0)

        ArducamSDK.Py_ArduCam_writeSensorReg(handle, 0x07, 0x398)

	if res == 0:

		openFlag = True

		print "device open success!"

		#while (regArr[regNum][0] != 0xFFFF):

			#ArducamSDK.Py_ArduCam_writeSensorReg(handle,regArr[regNum][0],regArr[regNum][1])

			#regNum = regNum + 1

		

		

		res = ArducamSDK.Py_ArduCam_beginCapture(handle)

		#print "Res value \n"

		#print  res

		#ArducamSDK.Py_ArduCam_writeSensorReg( handle, 0x46, 1)

		

		if res == 0:

			print "transfer task create success!"

			while flag :		

				#print "inside the while loop \n"

				res = ArducamSDK.Py_ArduCam_capture(handle)

				if len(data) >= Width * Height:

					if time1 - time0 >= 1:

						print "%s %d %s\n"%("fps:",count,"/s")

						count = 0

						time0 = time1

					show(data)

				

				if res != 0:

					print "capture fail!"

					break

				time.sleep(0.005)

				if flag == False:		

					break

		else:

			print "transfer task create fail!"

		res = ArducamSDK.Py_ArduCam_close(handle)

		if res == 0:

			openFlag = False

			print "device close success!"

		else:

			print "device close fail!"

	else:

		print "device open fail!"
`

Moinelly, 2018-02-05:
@ArduCAM 
Hey ,
Have you updated your so file.because when i run the code i am getting an import module error for
ArducamSDK.Py_ArduCam_available(handle) > 0
I have referred to your library which you shared one month back,in it the spell for available was availiable.
Regards,
vallabh