ArduCAM/ArduCAM_USB_Camera_Shield #14: SDK for ARM64 for c/c++

YurySams, 2017-12-15:
Hi,
We are looking a way to connect ARDUCAM USB SHIELD to our development board with arm64 cpu. 
We would use C/C++ language to process capturing video frame. 
Is it possible to share SDK (.so + header) with us?
thank you

ArduCAM, 2017-12-18:
Would you please let me know what kind of ARM64 CPU you are using?
The Linux SDK now we offer is only for PC and Raspberry pi, we need to work on other platform if needed.

alexanderhmw, 2018-02-20:
Hi,
I also have the similar problem. We need to deploy the ArduCAM on NVIDIA Tiger TX2, but we don't know how to compile the ArducamSDK.so. Could you please share your source code for building.

ArduCAM, 2018-02-22:
Do you have any email contact?

ryanmklee, 2018-03-12:
Hi,
Also having a similar problem, could you compile it in ARM64 or provide the source code? We're trying to use it with the NVIDIA Jetson TX2.

YurySams, 2018-03-14:
There is another solutions 
https://ccjt.taobao.com/?spm=2013.1.1000126.3.26d358c2zPDOl3

hardware, camera + usb shield, is good. But they provide software only for windows. However they share protocol and you can implement client basing on libusb for platform where you can find one, libusb.

lachinaj, 2018-03-16:
Hi,

I'm also interested to have it on ARM64 for NVidia Jetson TX1 and TX2.
I can help you to port it to this board.

Best regards

ArduCAM, 2018-03-29:
We have update the library for support ARM64 Arch processor, and tested on Pine64 board. Please check it out.

cullophid, 2018-04-24:
Have anyone had any luck making this work with the TX1/TX2?

luckyplay68, 2018-06-19:
Hi,
Is it possible to use the ArduCAM_USB_Camera_Shield with ODROID C2 board? Or ODROID XU4Q?
Can you update the library or share the source code?
Thank a lot.

ArduCAM, 2018-06-25:
Please try the library from: https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Pine64

luckyplay68, 2018-06-25:
Thanks a lot. We shall try it.

blubbi321, 2019-04-18:
@ArduCAM the link you posted is dead .. are there any plans to support the nvidia boards (tx2 and the like)?

ArduCAM, 2019-04-21:
Yes, we'd like to, but need to buy the tx2 board and port our library on that.

SK19, 2019-05-08:
Has anyone had any success getting these to work on a TX2?
I'm trying to set mine up with ROS but not having much luck currently.
Below is what happens when I attempt to launch it. Any suggestions on how to fix this?

[B_Camera_Shield/ROS/arducam_usb2_ros$ roslaunch arducam_usb2_ros arducam_node.launch
... logging to /home/nvidia/.ros/log/696d95fe-715b-11e9-b97e-00044ba7c041/roslaunch-tegra-ubuntu-27080.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:36800/

SUMMARY
========

CLEAR PARAMETERS
 * /cam0/

PARAMETERS
 * /cam0/arducam_ros_node/camera_serial: 
 * /cam0/arducam_ros_node/config_file: /home/nvidia/catk...
 * /cam0/arducam_ros_node/frame_id: cam0
 * /cam0/arducam_ros_node/horizontal_flip: True
 * /cam0/arducam_ros_node/vertical_flip: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /cam0/
    arducam_ros_node (arducam_usb2_ros/arducam_ros_node.py)

ROS_MASTER_URI=http://localhost:11311

process[cam0/arducam_ros_node-1]: started with pid [27098]
Traceback (most recent call last):
  File "/home/nvidia/catkin_ws/src/ArduCAM_USB_Camera_Shield/ROS/arducam_usb2_ros/src/arducam_ros_node.py", line 7, in <module>
    from ImageConvert import *
  File "/home/nvidia/catkin_ws/src/ArduCAM_USB_Camera_Shield/ROS/arducam_usb2_ros/src/ImageConvert.py", line 2, in <module>
    import ArducamSDK
ImportError: /home/nvidia/catkin_ws/src/ArduCAM_USB_Camera_Shield/ROS/arducam_usb2_ros/src/ArducamSDK.so: cannot open shared object file: No such file or directory
[cam0/arducam_ros_node-1] process has died [pid 27098, exit code 1, cmd /home/nvidia/catkin_ws/src/ArduCAM_USB_Camera_Shield/ROS/arducam_usb2_ros/src/arducam_ros_node.py __name:=arducam_ros_node __log:=/home/nvidia/.ros/log/696d95fe-715b-11e9-b97e-00044ba7c041/cam0-arducam_ros_node-1.log].
log file: /home/nvidia/.ros/log/696d95fe-715b-11e9-b97e-00044ba7c041/cam0-arducam_ros_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
](url)

ehong-tl, 2019-05-08:
> Has anyone had any success getting these to work on a TX2?
> I'm trying to set mine up with ROS but not having much luck currently.
> Below is what happens when I attempt to launch it. Any suggestions [...]

Try replace the ArducamSDK.so file in the ROS folder with the one in raspberrypi folder.

SK19, 2019-05-08:
If I do that, I get this outcome:
```
nvidia@tegra-ubuntu:~$ roslaunch arducam_usb2_ros arducam_node.launch
... logging to /home/nvidia/.ros/log/b2b255ca-715f-11e9-aa53-00044ba7c03f/roslaunch-tegra-ubuntu-2947.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:39115/

SUMMARY
========

CLEAR PARAMETERS
 * /cam0/

PARAMETERS
 * /cam0/arducam_ros_node/camera_serial: 
 * /cam0/arducam_ros_node/config_file: /home/nvidia/catk...
 * /cam0/arducam_ros_node/frame_id: cam0
 * /cam0/arducam_ros_node/horizontal_flip: True
 * /cam0/arducam_ros_node/vertical_flip: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /cam0/
    arducam_ros_node (arducam_usb2_ros/arducam_ros_node.py)

auto-starting new master
process[master]: started with pid [2957]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b2b255ca-715f-11e9-aa53-00044ba7c03f
process[rosout-1]: started with pid [2975]
started core service [/rosout]
process[cam0/arducam_ros_node-2]: started with pid [2980]
Traceback (most recent call last):
  File "/home/nvidia/catkin_ws/src/ArduCAM_USB_Camera_Shield/ROS/arducam_usb2_ros/src/arducam_ros_node.py", line 7, in <module>
    from ImageConvert import *
  File "/home/nvidia/catkin_ws/src/ArduCAM_USB_Camera_Shield/ROS/arducam_usb2_ros/src/ImageConvert.py", line 2, in <module>
    import ArducamSDK
ImportError: /home/nvidia/catkin_ws/src/ArduCAM_USB_Camera_Shield/ROS/arducam_usb2_ros/src/ArducamSDK.so: wrong ELF class: ELFCLASS32
[cam0/arducam_ros_node-2] process has died [pid 2980, exit code 1, cmd /home/nvidia/catkin_ws/src/ArduCAM_USB_Camera_Shield/ROS/arducam_usb2_ros/src/arducam_ros_node.py __name:=arducam_ros_node __log:=/home/nvidia/.ros/log/b2b255ca-715f-11e9-aa53-00044ba7c03f/cam0-arducam_ros_node-2.log].
log file: /home/nvidia/.ros/log/b2b255ca-715f-11e9-aa53-00044ba7c03f/cam0-arducam_ros_node-2*.log

```

ehong-tl, 2019-05-08:
@SK19 
It seems like the current SDK only support Linux PC and raspberrypi, have to wait for the developer to work on other platform like TX2.

ArduCAM, 2019-05-08:
We will order a TX2 or Jetson Nano to test it.
Please be patient.

JakubRN, 2019-05-08:
> We have update the library for support ARM64 Arch processor, and tested on Pine64 board. 

I'm also interested in integration with Jetson TX2 board, but if ARM64 support is there, I don't suppose it should be a problem, it is just not tested. I couldn't find any build though.
>Please check it out.
Could you in the meantime point me to where can I check it out?

insangi, 2019-05-31:
> We will order a TX2 or Jetson Nano to test it.
> Please be patient.

Have you tested this on Jetson Nano. Mine has the same problem I am getting an error ArducamSDK.so: cannot open shared object files: No such file or directory

But the file is there. Can you please share the source code so we can compile the SDK for the jetson Nano or any solution that can work with Nano. We have two USB sheilds 2.0 and 3.0 both are having same issue.

ArduCAM, 2019-06-03:
We just upload the new  update library to support Jetson Nano, it works for both USB2 and USB3 camera shield. Let me know if you have any problem.

Hackerman342, 2019-11-28:
Hi, I am also getting the error others have detailed above when I run the new package (cloned today) on a TX2. I have been working with the package on a linux laptop without issue for a couple months and only recently have tried running the package on TX2. 

Any advice would be greatly appreciated.
Thanks you.

"ImportError: /home/nvidia/svea_starter/src/ArduCAM_USB_Camera_Shield-master/ROS/arducam_usb2_ros/src/ArducamSDK.so: cannot open shared object file: No such file or directory"

glddiv, 2019-12-03:
Hi @Hackerman342 
Can you try to copy the shared library under the [Jetson Nano folder](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Nvidia_Jetson/Python/Streaming_demo) to the [ROS folder](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/ROS/arducam_usb2_ros/src)?
`cp ArduCAM_USB_Camera_Shield/Nvidia_Jetson/Python/Streaming_demo/ArducamSDK.so ArduCAM_USB_Camera_Shield/ROS/arducam_usb2_ros/src/`

ArduCAM, 2019-12-04:
We now moved our support to our forum, please resent the questions to https://www.arducam.com/forums/ 
Thank you.