ArduCAM/ArduCAM_USB_Camera_Shield #15: Trigger Mode for AR0134 in RPI

Moinelly, 2017-12-25:
Hey Lee,
So far everything is running perfectly,I need help in setting up trigger mode for AR0134 in RPI.
I have used your demo code for Auto-mode but when i change the register value in regArr of your code to trigger mode. i'm getting - capture fail error.
can you provide a sample code like to use trigger mode in RPI.
Thank you.
Please respond ASAP.

Moinelly, 2018-01-19:
@flyflyfly001 Yes,issue is solved now.There was problem in regArr value.Now everything is working good and as expected.
If you could provide me the code which you are using i can help you in the changes to make in the code to make it run in trigger mode.

flyflyfly001, 2018-01-19:
@Moinelly Hi, Moinely.
My Code is running on zedboard. It may different from ArduCAM. I set the 0x301A register value to 19D8. Is it correct?
Below is my reg config:
{0x3000, 0x0554},
		{0x3002, 0x0000},
		{0x3004, 0x0000},
		{0x3006, 0x03BF},
		{0x3008, 0x04ff},
		{0x300a, 0x03e5},
		{0x300c, 0x0672},
		{0x300e, 0x0023},
		{0x3010, 0xbeef},
		{0x3012, 0x002d},//integration time 64  2d
		{0x3014, 0x0000},
		{0x3016, 0x0010},
		{0x3018, 0x0000},
		{0x301a, 0x19d8},//19d8 enable trigger;   10dc disable trigger
		{0x301c, 0x0100},
		{0x301e, 0x00a8},
		{0x3026, 0x6500},
		{0x3028, 0x0010},
		{0x302a, 0x0008},//
		{0x302c, 0x0001},//
		{0x302e, 0x0008},//
		{0x3030, 0x0084},//0xc6 45fps      0x84  30fps
		{0x3032, 0x0000},
		{0x303a, 0x0258},
		{0x303c, 0x0000},
		{0x3040, 0x0000},
		{0x3044, 0x0400},
		{0x3046, 0x0000},
		{0x3056, 0x0020},
		{0x3058, 0x0020},
		{0x305a, 0x0020},
		{0x305c, 0x0020},
		{0x305e, 0x0020},
		{0x3064, 0x1982},
		{0x306e, 0x9000},
		{0x3070, 0x0000},

		{0x3072, 0x0000},
		{0x3074, 0x0000},
		{0x3076, 0x0000},
		{0x3078, 0x0000},
		{0x307a, 0x0000},
		{0x3086, 0x1700},
		{0x3088, 0x010a},
		{0x308a, 0x0002},
		{0x308c, 0x0004},
		{0x308e, 0x0501},
		{0x3090, 0x03c3},
		{0x30a0, 0x0001},
		{0x30a2, 0x0001},
		{0x30a4, 0x0001},
		{0x30a6, 0x0001},
		{0x30a8, 0x003f},
		{0x30aa, 0x005a},
		{0x30ac, 0x02a0},
		{0x30b0, 0x0080},
		{0x30b2, 0x0196},
		{0x30b4, 0x0000},
		{0x30bc, 0x0020},
		{0x30be, 0x0020},
		{0x30c0, 0x0020},
		{0x30c2, 0x0020},
		{0x30c4, 0x0020},
		{0x30c6, 0xf1b7},
		{0x30c8, 0x01bd},
		{0x30ca, 0xf69d},
		{0x30cc, 0x0200},
		{0x30d4, 0xe007},
		{0x3100, 0x0000},//auto exposure 1
		{0x3102, 0x0500},
		{0x3108, 0x0070},
		{0x310a, 0x0008},
		{0x310c, 0x0200},

		{0x310e, 0x2000},
		{0x3110, 0x0140},
		{0x311c, 0x02a0},
		{0x311e, 0x0001},
		{0x3124, 0x7fff},
		{0x312a, 0x0020},
		{0x3140, 0x0000},
		{0x3142, 0x0000},
		{0x3144, 0x0500},
		{0x3146, 0x03c0},
		{0x3152, 0x00f8},
		{0x3164, 0x02a0},
		{0x3166, 0x03da},
		{0x3168, 0x01a3},
		{0x3188, 0x0000},
		{0x31c0, 0x0000},
		{0x31c6, 0x8000},
		{0x31c8, 0xffff},
		{0x31ca, 0xffff},
		{0x31cc, 0xffff},
		{0x31ce, 0xffff},
		{0x31d2, 0x0000},
		{0x31d6, 0xdfcb},
		{0x31e8, 0x0000},
		{0x31ea, 0x0000},
		{0x31ec, 0x0000},
		{0x31ee, 0x0000},
		{0x31fc, 0x3020},
		{0x31fe, 0x0002},

Moinelly, 2018-01-19:
@flyflyfly001, set 0x301A register value to 0x1982

flyflyfly001, 2018-01-19:
@Moinelly Hi, Moinelly.
 I just tried. It makes no difference when I pull the trigger pin from low to high. I want to run in auto trigger mode. Any possible reason?

Moinelly, 2018-01-19:
@flyflyfly001  To make the camera run in Auto-Trigger mode set 0x301a to 0x10dc and to make the camera run in Trigger mode set 0x301a to 0x1982.
If you are using ARDUCAM AR0314 then first test it with the software of Arducam in windows later you can switch to zed board.

Moinelly, 2018-01-20:
@flyflyfly001 hey,
If you are using Arducam's AR0134 then there is a software in windows where you can test everything about camera including changing the register values.
What i suggest you is first test it on the windows version then you can test it on the zed board.

flyflyfly001, 2018-01-20:
@Moinelly  Hi, Moinelly.
I didn't have Arducam's AR0134. I'm trying to use pulse trigger mode to see if there has any new discovery. Thank you for you suggestion. Have a nice day!

Moinelly, 2018-01-20:
@flyflyfly001 According to AR0134 datasheet for external trigger set 0x301A to 0x1982.And minimum time between two triggers should be greater than exposure and frame valid time.
`
```
[
	#//[PLL_settings]
	[0x3028, 0x0010],		#//ROW_SPEED = 16
	[0x302A, 0x0010],		#//VT_PIX_CLK_DIV = 16
	[0x302C, 0x0001],		#//VT_SYS_CLK_DIV = 1
	[0x302E, 0x0002],		#//PRE_PLL_CLK_DIV = 2
	[0x3030, 0x0028],		#//PLL_MULTIPLIER = 40
	[0x3032, 0x0000],		#//DIGITAL_BINNING = 0
	[0x30B0, 0x0080],		#//DIGITAL_TEST = 128

	#//[Timing_settings]
	#[0x301A, 0x00D8],		#//RESET_REGISTER = 216
	[0x301A, 0x1982],		#//RESET_REGISTER = 4316
	[0x3002, 0x0000],		#//Y_ADDR_START = 0
	[0x3004, 0x0000],		#//X_ADDR_START = 0
	[0x3006, 0x03BF],		#//Y_ADDR_END = 959
	[0x3008, 0x04FF],		#//X_ADDR_END = 1279
	[0x300A, 0x0488],		#//FRAME_LENGTH_LINES = 1160
	[0x300C, 0x056C],		#//LINE_LENGTH_PCK = 1388
	[0x3012, 0x00D8],		#//COARSE_INTEGRATION_TIME = 216
	[0x3014, 0x00C0],		#//FINE_INTEGRATION_TIME = 192
	[0x30A6, 0x0001],		#//Y_ODD_INC = 1
	[0x308C, 0x0000],		#//Y_ADDR_START_CB = 0
	[0x308A, 0x0000],		#//X_ADDR_START_CB = 0
	[0x3090, 0x03BF],		#//Y_ADDR_END_CB = 959
	[0x308E, 0x04FF],		#//X_ADDR_END_CB = 1279
	[0x30AA, 0x0488],		#//FRAME_LENGTH_LINES_CB = 1160
	[0x3016, 0x00D8],		#//COARSE_INTEGRATION_TIME_CB = 216
	[0x3018, 0x00C0],		#//FINE_INTEGRATION_TIME_CB = 192
	[0x30A8, 0x0001],		#//Y_ODD_INC_CB = 1
	[0x3040, 0x4000],		#//READ_MODE = 0
	#//{0x3064, 0x1982},		#//EMBEDDED_DATA_CTRL = 6530
	[0x3064, 0x1802],
	[0x31C6, 0x8008],		#//HISPI_CONTROL_STATUS = 32776
	[0x301E, 0x0000],		#//data_pedestal
	#//{0x3100, 0x0001},		#//auto exposure

	[0x3002, 0],			#// Y_ADDR_START
	[0x3012, 150],
	
	[0x3056, 0x0024],		#// Gr_GAIN 
	[0x3058, 0x0028],		#// BLUE_GAIN
	[0x305a, 0x0028],		#// RED_GAIN 
	[0x305c, 0x0024],		#// Gb_GAIN 

	[0x3046, 0x0100],		#//en_flash/Flash indicator

	[0x301a, 0x1982],

	```` 

try this register settings.

flyflyfly001, 2018-01-20:
@Moinelly Hi, Moinelly.
I just used you reg config. The trigger mode worked well !! Thank you very much.

flyflyfly001, 2018-01-22:
@Moinelly Hi, Moinelly.
The pulse trigger mode is worked. How can I make sure that trigger pulse synchronize with the frame end signal?

smartexpt, 2018-05-17:
Hi @Moinelly and @flyflyfly001 ( @ryanmklee and @ArduCAM  as well)

I'm a novice in image sensors and I would like to integrate the Arducam with Raspberry Pi just to do the following steps:

1. Capture JPEG image (controllable exposure time);
2. Save image;
3. Wait a specified amount of time to return to step 1.

As there's little to no documentation available for ArduCAM I'm quite confused on where to start. Can you please point me in a right direction?

Much appreciated!

Moinelly, 2018-05-18:
@smartexpt there is code here in arducam repository to integrate AR0134 with RPI.
They have done in both CPP and python, you can try it.
And if you need help you can contact here anytime.

atfields43, 2018-09-10:
@Moinelly Hi I am trying to get the AR0134 to capture an image when externally triggered by a pi. I have used your register values, but I'm unsure how to implement the trigger mode. 

If I have R0x301a set to 0x1982 I should be in trigger mode, BUT:

How do I caputre an image when the device is triggered?
Do i have to manually check the trigger value, manually beginCapture, Capture, and endCapture, and then read it in? Or does it automatically place a picture in the imageAvailable buffer when triggered, and all I have to do is read it? 

I can't find info on this for the raspberry pi.

Moinelly, 2018-09-11:
@atfields43 Hey there,
if you have set 0x301a to 0x1980 the camera should be in trigger mode and apply 5v trigger to the given hardware pin.
when the trigger is generated it automatically captures the frame and we can read it using Arducam_read function. you don't have to check manually for the trigger value. 
Use there demo code to test it, it works perfectly.
By the way which version of USB shield are you using now, is it USB shield version 1 or version 2.

atfields43, 2018-09-11:
@moinelly thanks for the reply! so I enabled the "FLASH" output, which sycrhonizes the exposure time, and I see that the exposure time occurs right after the trigger and for the correct time I manually set. however, no image is available using the arducam_imageavailable afterward and the arducam_readimage does not return that a picture was read. where is there a demo code for trigger mode? I downloaded the zip and there's only demo code for streaming. USB 2.0

Moinelly, 2018-09-11:
@atfields43 firstly i would like to say that implement the same in windows using there Demo GUI
[https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Windows/GUI/USBTest](url)
and test the trigger mode in there.
It will get you an idea how it is working.

Thank you!

atfields43, 2018-09-11:
@Moinelly I have tested the trigger mode on the GUI. Same thing. I know its exposing because I see the flash output high for the duration for the exposure, but no where on the GUI is there a way to display the image. If I press play, its expecting to be in auto mode, so it gives me a USB timeout error, if I press shot, nothing happens, I changed the mode register to the correct thing, but neither on python or the GUI can i extract the data that is obviously being created. 
`GPIO.output(HIGH)
time.sleep(0.001)
GPIO.output(LOW)
if ArduCam_imageAvailable() > 0:
     ArduCam_readImage(handle)
else:
     print "no image"
'
this always prints "no image" even though I see that the camera was exposed. I will try to work on it but I don't understand why I'm having such trouble with such a simple exercise.

ArduCAM, 2018-09-11:
@atfields43 Please send us your modified code, we will try to test that tomorrow.

atfields43, 2018-09-11:
@ArduCAM I sent the code to admin@arducam.com over email. Thank you.

ArduCAM, 2018-09-11:
OK , I have got your code, and will work on that tomorrow, please be patient.

erikbeall, 2018-09-18:
I'm adding to this topic because I have a related issue, JPEG mode on AR0134. Is jpeg compression done on the USB board or is it only supported if the image sensor chip supports it? It would be useful for reducing USB bandwidth for those of us who are constrained.

ArduCAM, 2018-09-18:
AR0134 doesn't support JPEG mode and USB board can't do JPEG either. You can only save JPEG image with OpenCV function.

erikbeall, 2018-09-18:
Ah, so it goes. Thanks.