ArduCAM/ArduCAM_USB_Camera_Shield #17: Capture in continous mode

falcald, 2018-01-22:
Hi, let me tell you first that your device is awesome to learn and handle image sensors in a fast and easy way. So congratulations!

Now, in my application I'm using the MT9M001 Aptina sensor and I would like to capture frames with a very long integration time (max is 1033ms) but without losing any frame. The issue is that, in your example code, calling to `ArducamSDK.Py_ArduCam_capture(handle)` apparently waits for the next frame before recording the data, and so by polling it you capture one frame and always lose the next. What I'm looking for is a way to continuously capture every frame output by the sensor without having to call a "capture" method everytime. Something like:

Thread 1:
```
ArducamSDK.Py_ArduCam_beginCapture(handle); 
# That would start a thread that continuously read and save data to framebuffer
while 1:
   DoNothing();
```
Thread 2:
```
while 1:
   if ArducamSDK.Py_ArduCam_availiable(handle) > 0:
      # Read and process data.
      res,data = ArducamSDK.Py_ArduCam_read(handle,Width * Height) 
```

The USB bandwidth shouldn't be a problem here because as I said the integration time is very long and so the data rate is very low (1fps).

Hope you can help me with this. Thanks in advance and thank you again for this very good device.

Moinelly, 2018-02-05:
@falcald Have you found the solution,for this problem?

falcald, 2018-02-05:
@Moinelly No, sorry. I don't have the ArducamSDK library code nor the board firmware to fix this. The developers should take a look at this issue.