ArduCAM/ArduCAM_USB_Camera_Shield #20: Regarding get他、ing raw

ZuoJiawei, 2018-02-21:


ZuoJiawei, 2018-02-21:
Is it possible to read raw ADC value from each camera pixel and save the raw data as .mat or .txt file?

ArduCAM, 2018-02-22:
If you are using RAW sensor, it can save the raw data as .bin blob file, you can convert the .bin to any other format yourself.

ZuoJiawei, 2018-02-22:
Yes,I'm trying to use raspberry pi 3 to read ArduCAM module MT9J001. Inspired by your reply,shall the cmd line below works for reading raw ADC data?
sudo python MT9J001.py --type bin --name sample_image

ArduCAM, 2018-02-22:
You can try the camera on windows or linux ubuntu first before moving to RPI.
www.arducam.com/downloads/app/ArduCAM_USB/x86/ArduCAM_Linux_SDK_OpenCV_MT9J001_Demo.zip
We will update another C/C++ version for RPI very soon.