ArduCAM/ArduCAM_USB_Camera_Shield #27: Visible banding in images from MT9J001

rtb21, 2018-03-28:
Hi,

I've just purchased a MT9J001 (monochrome) camera and the ArduCAM USB Camera Shield and in general it is working, however the images I'm receiving from the camera have a couple of issues.
Firstly the images (especially in low light conditions) contain significant horizontal structure, and secondly the images also exhibit banding in the light level (the wall next to the light should be brighter than its surrounding but appears darker)
[image]

Thanks,
Rob

ArduCAM, 2018-03-29:
Rob
Please try to increase the global gain register 0x0204 value like 0x80, and you can change the exposure register 0x3012 for much brighter image.
Let me know your testing result.

ArduCAM, 2018-03-30:
Would you please reseat the ribbon cable on both end of the camera and the usb board, sometimes the cable tend to be loose. You can also send me your order number to admin@arducam.com, so if problem still exist, we can send you a replacement ASAP.
You can find the reigster reference from: www.arducam.com/downloads/modules/industrial/MT9J003_RR_A.pdf   
Let me know if you need more help.

ArduCAM, 2018-04-03:
I still didn't get your reply, just waiting for your kind reply.

rtb21, 2018-04-03:
Sorry for the delay, I have been away for a few nights.

The good news is that the camera now appears to be working, however I all I did was retry it again tonight, so I don't know what happened to fix the issue! Still, thanks for the quick responses and the link to the register reference.

Rob

closed 2018-04-03 by rtb21