ArduCAM/ArduCAM_USB_Camera_Shield #29: Camera shield USB data exchange format

RenGate, 2018-05-01:
Hi Lee, first of all big thanks to you for supporting the code base in this repository. Right now the SDK itself is closed source, which prohibits people from porting the software to other platforms or programming languages (as was mentioned in #13). 
If open sourcing Arducam SDK code is not an option, can you please give us at least some info on how data is parsed by the SDK? What I'm particularly curious about is how much data must be transferred from the shield using USB bulk transfers to get 1 frame and what is its structure, i.e. how raw pixel data can be extracted from it? Thanks.

ArduCAM, 2018-05-02:
Please contact us by admin@arducam.com for detail information.

RenGate, 2018-05-10:
Lee kindly answered all my questions over an email. I will summarize the main points here in case if anyone else is interested.
* The image data is transmitted from USB camera shield to the PC using USB bulk transfers. The size of bulk transfer must be a multiple of 512 bytes.
* To separate individual frames a marker is used. It begins with a 10 byte sequence of [1, 2, 3, 4, 5, 6, 7, 8, 9, 170].
* The image data is split into the 512 byte blocks. If data length isn't multiple of 512, not all data in the last data block will be valid.
* There is no frame buffer on the USB camera shield, hence the data readout must be performed as quick as possible. Asynchronous libusb functions for bulk transfers must be used.

closed 2018-05-10 by RenGate