ArduCAM/ArduCAM_USB_Camera_Shield #30: Snapshot unavailable for AR0134

smartexpt, 2018-05-17:
Hey guys,

Congrats on this amazing project, getting image sensors available for everyone!

This issue reports to the fact that, as following the steps on the `/ArduCAM_USB_Camera_Shield/Linux_x86/Python/README.md` file, the snapshot function is not working.

When I type:

`sudo python AR0134.py --type jpg --name sample_image` 

I get the following error:

```
usage: AR0134.py [-h] [-i TYPE]
AR0134.py: error: unrecognized arguments: --name sample_image
```

I tried to work it around by adding the `name` argument as `ap.add_argument("--name",required = False, help = "name to the image file")` and modifying the `show(bufferData)` to:

```
def show(bufferData):
	global downFlag,saveFlag,saveNum
	image = Image.frombuffer("L",(Width,Height),bufferData)
	img = np.array(image)
	height,width = img.shape[:2]
	img2 = cv2.cvtColor(img,COLOR_BYTE2RGB)
        saveNum += 1
	
	if args["name"] != "":
                name = args["name"]   
        else:
                name = str(saveNum)
	if "bmp" == args["type"]:        
                name += ".bmp"
        if "png" == args["type"]:        
                name += ".png"
        if "jpg" == args["type"]:        
                name += ".jpg"
	cv2.imwrite(name,img2)	
```

That way, it actually saves the snapshot but the program and logging to the terminal.

Final question is: how can I get a simple snapshot, NOT in video mode but as an actual picture?

PS: It would be a major upgrade if you guys could provide more documentation and maybe the source code for the SDK

UCTRONICS, 2018-05-21:
@smartexpt 
Sorry for reply you so late because of the weekend.
The  operation  of snapshot function is the same as  the stream. If you just want to get an image ,
you just need to close the program after you get the correct picture in read Thread.
Please try this python demo I attachd for you.
[AR0134_snapShot.zip](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/files/2021123/AR0134_snapShot.zip)

UCTRONICS, 2018-05-22:
@smartexpt 
Just run 'sudo python  AR0134_snapShot.py -c -n picturename -s' to get a picture.
"-c" means take a photo,"-n" is the name of the picture,"-s" means that this picture is displayed, you can not use it.