ArduCAM/ArduCAM_USB_Camera_Shield #32: ArduCam_open failed

SanggunLee, 2018-05-23:
I received 'ArduCAM_USB_Camera_Shiled' and 'AR0134 RGB camera module'.

On my ubuntu 16.04(64bit), I built 'ArduCAM_USB_Camera_Shield/Linux_x86/Cpp/ArduCAM_Linux_SDK_OpenCV_AR0134_Demo/AR0134_demo.cpp'.

And I ran and got error message 'Device found = 1   ArduCam_open failed'.

AR0134_demo.cpp, Line 144
u32Value = ArduCam_autoopen( useHandle, &useCfg );

This function doesn't work.

What is a problem?
Is it possible that camera module is broken?
Or, ArduCam_autoopen has bug...?

How can I check??

ArduCAM, 2018-05-23:
You have to use sudo before the command.

SanggunLee, 2018-05-24:
Thank you. Working well :)

UberEclectic, 2019-04-20:
Out of the box the /dev/bus/usb/???/??? files will be owned by root:root with perms 644, like this:

ls -l /dev/bus/usb/001/00*
crw-rw---- 1 root root 189, 0 Apr 20 14:02 /dev/bus/usb/001/001
crw-rw---- 1 root root 189, 1 Apr 20 14:02 /dev/bus/usb/001/002
crw-rw---- 1 root root 189, 2 Apr 20 14:02 /dev/bus/usb/001/003
crw-rw---- 1 root root 189, 3 Apr 20 14:12 /dev/bus/usb/001/004
crw-rw---- 1 root root 189, 4 Apr 20 14:02 /dev/bus/usb/001/005

What I did to avoid running under sudo/root was add a udev rule like this:

$ cat /etc/udev/rules.d/99-usb.rules
SUBSYSTEM=="usb", GROUP="input", MODE="0660"

In the above example, I used group "input" because I can see (using "id -a") that my non-root users in already a member of group "input" -- you can vary the group to meet your needs.

Now I see:

$ ls -l /dev/bus/usb/001/00*
crw-rw---- 1 root input 189, 0 Apr 20 14:02 /dev/bus/usb/001/001
crw-rw---- 1 root input 189, 1 Apr 20 14:02 /dev/bus/usb/001/002
crw-rw---- 1 root input 189, 2 Apr 20 14:02 /dev/bus/usb/001/003
crw-rw---- 1 root input 189, 3 Apr 20 14:12 /dev/bus/usb/001/004
crw-rw---- 1 root input 189, 4 Apr 20 14:02 /dev/bus/usb/001/005

And with the above, I do not need to run the binary under root/sudo.

closed 2021-06-02 by SanggunLee