ArduCAM/ArduCAM_USB_Camera_Shield #35: AR0134 USBtest image size changing

frascuelillo, 2018-05-30:
How can I show an image of a size different than 1280x964 ?, when I enter another value no image is shown, and if I select the option to force display, the image shown is corrupt.
[image]

ArduCAM, 2018-05-30:
Actually for AR0134  there is extra 4 default lines enabled, so please input 484 in the Height edit box to match with your setting.

frascuelillo, 2018-05-31:
Thank's, 
But I don't have success with that setting:

[image]

ArduCAM, 2018-05-31:
Can you send me your settings, so I can figure this out here.

frascuelillo, 2018-06-04:
Hi lee, 
I sended you an email with my settings.
It´s preferable paste my settings in this post?

ArduCAM, 2018-06-04:
Sorry, just sent by email to you.

closed 2018-06-09 by frascuelillo