ArduCAM/ArduCAM_USB_Camera_Shield #36: ROS integration

tiralonghipol, 2018-06-04:
Hi !
I would like to know: is there any already-existing ros integration library or package for the arducam shield?
If yes, do you have any link?
Thanks!

ArduCAM, 2018-06-04:
Sorry we don't have that right now. Would you please contact us for customized library cooperation. admin@arducam.com

mzahana, 2018-08-24:
Hi. I am also interested in a ROS integration.  It would be more beneficial to have it as it can facilitate the integration in many robotics projects.

ArduCAM, 2018-08-26:
send me email to discuss the integration to ROS.

tiralonghipol, 2018-08-27:
Due to some problems at work I could not make any improvements on this
If you need any help feel free to mail me at tiralonghipol@gmail.com

nifunk, 2018-08-31:
Hey all, I would also be very interested in  a ROS integration. Did anyone of you sucessfully implement an integration yet? Is there any starting point on which I could build upon?

tiralonghipol, 2018-08-31:
@nifunk  we were about to plan on how to procede..
In my opinion, we should open a new branch on this repo, trying to modify the ArduCam_Py_Demo.py
It is really a shame the this camera is not ros-integrated yet!

ehong-tl, 2018-09-07:
Hey all, I've created a python node to run arducam in ROS.

https://github.com/ehong-tl/arducam_usb2_ros

Please let me know if there's any problem. :-)

tiralonghipol, 2018-09-07:
Hi
I was trying to roslaunch the node but it says
'can't locate node [arducam_ros_node.py] in package [arducam_usb2_ros]

How can this be solved?

Paolo

________________________________
WPWEB s.r.l.
Paolo De Petris
Via Livorno 60 - 10144 Torino
Cell +39.328.357.1370
<mailto:falcone@bmedia.it>
paolo.depetris@wpweb.com
<mailto:falcone@bmedia.it>
<mailto:falcone@bmedia.it>
http://www.wpweb.com<http://www.wpweb.com/>

A: ArduCAM/ArduCAM_USB_Camera_Shield
Cc: Paolo De Petris; Author

https://github.com/ehong-tl/arducam_usb2_ros

ehong-tl, 2018-09-07:
Hi Paolo,

You need to allow permission for the python file to run as program.

This can be fix by 'chmod 777 arducam_ros_node.py'

tiralonghipol, 2018-09-07:
Also, do you think it is possibile to add some parameters? like exposure / gain / frame rate?

________________________________
WPWEB s.r.l.
Paolo De Petris
Via Livorno 60 - 10144 Torino
Cell +39.328.357.1370
<mailto:falcone@bmedia.it>
paolo.depetris@wpweb.com
<mailto:falcone@bmedia.it>
<mailto:falcone@bmedia.it>
http://www.wpweb.com<http://www.wpweb.com/>

Added to readme file :-)

-

ehong-tl, 2018-09-07:
For this, we might need help from @ArduCAM , cause the SDK user guide provided is somewhat limited for controlling Arducam's parameters like exposure, gain, etc.

ehong-tl, 2018-09-07:
Hi @tiralonghipol ,

I believe you can adjust the parameter by editing the json config file for your camera.

closed 2018-09-07 by tiralonghipol

ehong-tl, 2018-09-10:
Hi @tiralonghipol ,

I've added service server to allow read/write value from/to sensor register, which allows you to manually adjust exposure, gain, etc.

https://github.com/ehong-tl/ArduCAM_USB_Camera_Shield/tree/ROS-support/ROS/arducam_usb2_ros

Please check it out and let me know if there's any issue.

I've pulled a request to merge with the master repo.