ArduCAM/ArduCAM_USB_Camera_Shield #37: Image loading error.

youn-seong-rim, 2018-06-06:
I want to transfer images to PC using arducam usb camera shield.
The software I am using is usbtest.exe, a window test program provided by arducam.
And the hardware I am using is arducam usb camera shield and ov5642 camera module.
The usb driver is recognized in the device manager, and when the open button is clicked on the usbtest program, the message "usb open succeeded" appears.
However, when I click the play button, the image is not shown and only the time out error message is displayed.
So I am doing hardware debugging and I have found one problem.
The problem is that the source clock is not oscillating from the FPGA of the arducam usb camera shield to the pin 8 (XCLK) of the camera module.
I do not know if the board is a problem or if I have misconfigured it or if I have another problem.
Let me know the cause and solution of the above problem.

ArduCAM, 2018-06-06:
You have to start the camera using usbtest.exe then using oscillosocpe to watch the XCLK clock. If you don't start the usbtest.exe, the XCLK will stop after few seconds. 
I suggest you to use usbtest.exe initialize the camera first, then use oscilloscope to watch the PCLK, VSYNC, HREF to see if there is any waveforms, matter you received time out error message from the GUI.
Please send me the result, then I can give you more instructions to move to next steps.
Thanks.

youn-seong-rim, 2018-06-07:
Thank you for your prompt reply. As you answered, I ran usbtest.exe and clicked the open button, "usb camera init success" appeared in the message box, and I could confirm that the clock was generated by the xclk pin. But the image still does not come out. After checking with the oscilloscope, Hsync appears at the same cycle as Vsync. So, I expect that there is no change in the waveform of Hsync in the VSync cycle and that the value of data [7..0] can not be read by pclk.
I am trying to load ov5640.cfg file because there is no ov5642.cfg file which is a camera config file. Can this happen because this config file is not correct?
If so, is there any way to get or create an ov5642.cfg file?
I would appreciate your reply again.

ArduCAM, 2018-06-07:
OK, I know what's the problem. OV5640 configuration file can't work for OV5642. I will update the OV5642 config file soon.

ArduCAM, 2018-06-07:
Please check the  update config file for OV5642 : https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/blob/master/Windows/USBTest/x64/Release/Config/OV5642_JPEG_QSXGA.cfg

youn-seong-rim, 2018-06-08:
Leave a comment[](url)

closed 2018-06-08 by youn-seong-rim

vintagefilmography, 2019-06-09:
I have exactly same issue with the MT9P001 and usb camera shield. Please advice.