ArduCAM/ArduCAM_USB_Camera_Shield #39: Using Arducam USB Shield with Python under Windows.

frascuelillo, 2018-06-11:
It's possible to use Arducam USB shield in Python in a Windows PC?

ArduCAM, 2018-06-11:
Sorry, we didn't have Python for Windows now, only Linux available. But we now have opencv C++ version for windows, does that help?

frascuelillo, 2018-06-11:
Did you mean an OpenCV example? yes, it helps, even in c++ version.

ArduCAM, 2018-06-11:
Let me know what kind of camera module you are using, then I can upload a c++ version opencv example soon. And let me know your email address as well.

frascuelillo, 2018-06-11:
the module is AR0134, I will send you an email

ArduCAM, 2018-06-11:
Just sent you by email.

frascuelillo, 2018-06-15:
Thank´s Lee, 
I was feeling more comfortable in Python, but it's a great oportunity to learning C++. 
The example you send me work's fine.

closed 2018-06-15 by frascuelillo