ArduCAM/ArduCAM_USB_Camera_Shield #40: Questions about "Bad frame received" error

youn-seong-rim, 2018-06-18:
I would like to know about the following cases when using the usbtest program.
When I click the play button, only "Bad frame received" appears in the message box and nothing appears in the image window. And when I check the Force Display button, the image appears in the image window.
But the image seems to be unstable.I want to know why this is happening and how to solve it.

ArduCAM, 2018-06-18:
Please send me the cfg file, so I can check it out.

youn-seong-rim, 2018-06-19:
The cfg file I used is the ov7670 vga file.
It is difficult to attach the file, so I wrote down the contents of the config below.
Please check the contents below and reply.

[camera parameter]
CFG_MODE  = 0	
TYPE      = OV7670
SIZE      = 640, 480
BIT_WIDTH = 8 
FORMAT    = 1, 0	
I2C_MODE  = 0					
I2C_ADDR  = 0x42					

[board parameter]

[board parameter][dev2]
VRCMD = 0xD7, 0x4600, 0x0100, 1, 0x05
VRCMD = 0xF6, 0x0000, 0x0000, 3, 0x03, 0x04, 0x0C

[board parameter][dev3][inf2]
VRCMD = 0xF3, 0x0000, 0x0000, 0
VRCMD = 0xF9, 0x0000, 0x0000, 0

[board parameter][dev3][inf3]
VRCMD = 0xF3, 0x0000, 0x0000, 0           
VRCMD = 0xF9, 0x0000, 0x0000, 0

[register parameter]
          
REG = 0x3a, 0x04
REG = 0x40, 0xd0
REG = 0x12, 0x04 //output format //0x04(RGB565), 0x00(Yuv)
REG = 0x32, 0x80 //Href  default = 0x80, setting = 0x80
REG = 0x17, 0x16 //Hstart = Value + Href[2:0] default = 0x11, setting = 0x16
REG = 0x18, 0x04 //Hend = Value + Href[5:3] default = 0x61, setting = 0x04
REG = 0x19, 0x02 //Vstart = Value + Vref[1:0] default = 0x03, setting = 0x02
REG = 0x1a, 0x7b //Vend = Value + Vref[3:2] default = 0x7B, setting = 0x7b , 0x3f 
REG = 0x03, 0x06 //Vref  default = 0x03, setting = 0x06
REG = 0x0c, 0x00
REG = 0x3e, 0x00
REG = 0x70, 0x00
REG = 0x71, 0x00
REG = 0x72, 0x11
REG = 0x73, 0x00
REG = 0xa2, 0x02
REG = 0x11, 0x82 //pclk divider
REG = 0x7a, 0x20
REG = 0x7b, 0x1c
REG = 0x7c, 0x28
REG = 0x7d, 0x3c
REG = 0x7e, 0x55
REG = 0x7f, 0x68
REG = 0x80, 0x76
REG = 0x81, 0x80
REG = 0x82, 0x88
REG = 0x83, 0x8f
REG = 0x84, 0x96
REG = 0x85, 0xa3
REG = 0x86, 0xaf
REG = 0x87, 0xc4
REG = 0x88, 0xd7
REG = 0x89, 0xe8
REG = 0x13, 0xe0
REG = 0x00, 0x00
REG = 0x10, 0x00
REG = 0x0d, 0x00
REG = 0x14, 0x28
REG = 0xa5, 0x05
REG = 0xab, 0x07
REG = 0x24, 0x75
REG = 0x25, 0x63
REG = 0x26, 0xA5
REG = 0x9f, 0x78
REG = 0xa0, 0x68
REG = 0xa1, 0x03
REG = 0xa6, 0xdf
REG = 0xa7, 0xdf
REG = 0xa8, 0xf0
REG = 0xa9, 0x90
REG = 0xaa, 0x94
REG = 0x13, 0xe5
REG = 0x0e, 0x61
REG = 0x0f, 0x4b
REG = 0x16, 0x02
REG = 0x1e, 0x17
REG = 0x21, 0x02
REG = 0x22, 0x91
REG = 0x29, 0x07
REG = 0x33, 0x0b
REG = 0x35, 0x0b
REG = 0x37, 0x1d
REG = 0x38, 0x71
REG = 0x39, 0x2a
REG = 0x3c, 0x78
REG = 0x4d, 0x40
REG = 0x4e, 0x20
REG = 0x69, 0x00
REG = 0x6b, 0x00
REG = 0x74, 0x19
REG = 0x8d, 0x4f
REG = 0x8e, 0x00
REG = 0x8f, 0x00
REG = 0x90, 0x00
REG = 0x91, 0x00
REG = 0x92, 0x00
REG = 0x96, 0x00
REG = 0x9a, 0x80
REG = 0xb0, 0x84
REG = 0xb1, 0x0c
REG = 0xb2, 0x0e
REG = 0xb3, 0x82
REG = 0xb8, 0x0a
REG = 0x43, 0x14
REG = 0x44, 0xf0
REG = 0x45, 0x34
REG = 0x46, 0x58
REG = 0x47, 0x28
REG = 0x48, 0x3a
REG = 0x59, 0x88
REG = 0x5a, 0x88
REG = 0x5b, 0x44
REG = 0x5c, 0x67
REG = 0x5d, 0x49
REG = 0x5e, 0x0e
REG = 0x64, 0x04
REG = 0x65, 0x20
REG = 0x66, 0x05
REG = 0x94, 0x04
REG = 0x95, 0x08
REG = 0x6c, 0x0a
REG = 0x6d, 0x55
REG = 0x6e, 0x11
REG = 0x6f, 0x9f
REG = 0x6a, 0x40
REG = 0x01, 0x40
REG = 0x02, 0x40
REG = 0x13, 0xe7
REG = 0x15, 0x02 //clk reverse
REG = 0x4f, 0x80
REG = 0x50, 0x80
REG = 0x51, 0x00
REG = 0x52, 0x22
REG = 0x53, 0x5e
REG = 0x54, 0x80
REG = 0x58, 0x9e   
REG = 0x41, 0x08
REG = 0x3f, 0x00
REG = 0x75, 0x05
REG = 0x76, 0xe1
REG = 0x4c, 0x00
REG = 0x77, 0x01
REG = 0x3d, 0xc2  
REG = 0x4b, 0x09
REG = 0xc9, 0x60
REG = 0x41, 0x38
REG = 0x56, 0x40
REG = 0x34, 0x11
REG = 0x3b, 0x02
REG = 0xa4, 0x89
REG = 0x96, 0x00
REG = 0x97, 0x30
REG = 0x98, 0x20
REG = 0x99, 0x30
REG = 0x9a, 0x84
REG = 0x9b, 0x29
REG = 0x9c, 0x03
REG = 0x9d, 0x4c
REG = 0x9e, 0x3f
REG = 0x78, 0x04   
REG = 0x79, 0x01
REG = 0xc8, 0xf0
REG = 0x79, 0x0f
REG = 0xc8, 0x00
REG = 0x79, 0x10
REG = 0xc8, 0x7e
REG = 0x79, 0x0a
REG = 0xc8, 0x80
REG = 0x79, 0x0b
REG = 0xc8, 0x01
REG = 0x79, 0x0c
REG = 0xc8, 0x0f
REG = 0x79, 0x0d
REG = 0xc8, 0x20
REG = 0x79, 0x09
REG = 0xc8, 0x80
REG = 0x79, 0x02
REG = 0xc8, 0xc0
REG = 0x79, 0x03
REG = 0xc8, 0x40
REG = 0x79, 0x05
REG = 0xc8, 0x30
REG = 0x79, 0x26
REG = 0x09, 0x03
REG = 0x3b, 0x42
REG = 0xff, 0xff

ArduCAM, 2018-06-25:
Can you share your email address, so I can send you the correct settings.

youn-seong-rim, 2018-06-28:
My email address is yysung@naver.com.

bwbeall, 2018-06-29:
Would you please provide the settings for everyone?

ArduCAM, 2018-06-29:
Yes, we will publish that on github until we proof it works with whom has issue.

ArduCAM, 2018-06-30:
Please try this setting for OV7670
www.arducam.com/downloads/app/ArduCAM_USB/Config/OV7670_VGA_RGB565.cfg

bwbeall, 2018-06-30:
Thank you Lee!How are things? I hope everything is going well. Please let me know if I can do anything to help.Brook

ArduCAM, 2018-06-30:
extremely busy as usual, I think I might go insane.

simbioza, 2018-11-01:
I have similar problem with AR0134 on usb2.0 shield rev.b. Windows test program reads about 20 fps and about 4-5 bad fps. drivers and config file is up to date form this github. Any Ideas?

ArduCAM, 2018-11-01:
It is known issue. For Rev.B board there is no frame buffer, so when bandwidth rush, it will cause drop frame issue. so you have to reduce the frame rate in order to avoid the dropping frames. Our new USB2 camera shield Rev.C or Rev.D add extra frame buffer which helps to avoid such issues.