ArduCAM/ArduCAM_USB_Camera_Shield #42: Config file loading fail in my program

youn-seong-rim, 2018-06-28:
I want to use USBTEST source in my program. However, when I run the ArduCamCfg_LoadCameraConfig function, it fails to load the config file.
In particular, I ran the same code in usbtest and my program as in the code below.
In usbtest, the file was loaded without any problems. But I do not know why my code does not load files. Please give me advice.

csTmpFileName = _T ("D: \\ Config \\ AR0134_960p.cfg");
if (CFG_NONE_CONTENT == ArduCamCfg_LoadCameraConfig (m_cUsbCameraCfgHd, csTmpFileName))
{
AfxMessageBox (_T ("Config file Open failed!"));
return;
}

youn-seong-rim, 2018-06-30:
I'm sorry, but I will ask you again in detail.
I created a new project in visual studio (visual c ++) and wrote the code by referring to usbtest source code.
I ran the build without the compile error, and the program ran normally until the usb device open.
However, when I run the next step ArduCamCfg_LoadCameraConfig (.., ..), I get an abnormal return value (CFG_NONE_CONTENT). In other words, I can not load the config file properly. I have the same as the usbtest source code and no compile error, but I do not know why not.
Please answer or advise.

ArduCAM, 2018-06-30:
I suggest you follow the new example using opencv for Windows, it is much easier to understand 
www.arducam.com/downloads/app/ArduCAM_USB/x86/ArduCam_opencv_VS2017.zip

youn-seong-rim, 2018-07-04:
I have found the cause of the problem I asked above.
The reason is that new programs use Unicode character sets.
On the other hand, the USBTEST program used a multibyte character set.
As a result, it is possible to capture images without problems when using multibyte, but the above problem occurs when using Unicode. I would like to know why the above function causes problems when using Unicode. Please answer or advise.

ArduCAM, 2018-07-05:
Sorry, we only use multibyte character set now. You can't use Unicode character set.

youn-seong-rim, 2018-07-06:
Okay. Thank you for answer
But do not you have plans to upgrade to be available on Unicode?
I hope I can use it on Unicode if you can. Thank you.

ArduCAM, 2018-07-06:
Like I said please use the new demo for opencv from: www.arducam.com/downloads/app/ArduCAM_USB/x86/ArduCam_opencv_VS2017.zip 
We don't suggest to use the MFC USB Test demo in the future.

youn-seong-rim, 2018-07-12:
I Knoww what you mean. Thank you for answer.

ArduCAM, 2018-07-12:
Now we updated the example here : https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Windows/OpenCV