ArduCAM/ArduCAM_USB_Camera_Shield #43: Windows Driver (Code 48) Error

samerneergaard, 2018-07-10:
[image]

I've just got the AR0134 (global shutter) along with a USB2 ARDUCAM module and I've been having serious issues with it all day. 

1. I've tried setting it up with windows, manually installing the unsigned driver (already disabled the security on reboot) and its gives me a "Code 48" Windows blocks the driver completely kind of error.
NOTE: I have disabled the signed drivers, that is NOT the problem

2. I've installed Ubuntu on a VMbox and tried accessing the camera through python and cpp but the device is still not recognized 

3. Lastly, I've plugged in my raspberry pi 3 and set everything up successfully. But I get to the point where the python code runs successfully, then immediately fails. (closed because of high data)

Ultimately, the driver needs to be updated I think. But is there any way to get this in live preview for a test I'm running, I would greatly appreciate it.

Thanks

ArduCAM, 2018-07-10:
It looks like you still didn't install the driver properly. You have to follow the guide exactly from https://youtu.be/gOTkrFp8oM4 , although I know you will feel frustrated.
Please do not use VMbox or similar, please use real Ubuntu machine.
If you want to try RPI, please try the C++ example first.
Or please send us lsusb command information.

samerneergaard, 2018-07-10:
Hey,

Thanks for the quick reply. I’ve followed these tutorials as well as done some work in the cmd lines to permanently disable driver signatures. It seems that it’s something else.

In terms of using the cpp method. I’m embarrassed to say I was stuck on the moving the Arducamlib.so file to the /lib. That step was not very clear to me to be honest.

ArduCAM, 2018-07-10:
You can disable the signatures as the video shows for one time when booting up for try first, making sure the camera is no problem. Then we can narrow down the other software setting problems.

For the library file, I did mentioned in the readme file here： https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/blob/master/RaspberryPi/Cpp/README.md

Does it work with RPI now?

samerneergaard, 2018-07-10:
I only want to get this working. Ideally on windows. But this is definitely doesn’t seem to be working on my windows 10 because of the code 48 error. I’ve tried on two different computers Incase.

ArduCAM, 2018-07-10:
Maybe you have to find another Windows7 machine to try.
Sorry I didn't mention how do the copy command, I will update it now.
Our latest USB2 camera shield has onboard framebuffer which definitly help improve the performance on RPI, see our website.

samerneergaard, 2018-07-10:
Yes, I’ve done a lot of research. I’m using the new usb2 board from your website. Thanks for helping with raspberry pi.

As for windows it’s much more preferable to use newer versions of windows for this project. I’ve purchased this usb module based on that spec and ease of use but so far there have been issues with the drivers. Is there a way you can think of to update the drivers or bypass the security other than disabling unsigned drivers on boot that will solve this problem?

ArduCAM, 2018-07-10:
Can you setup a team viewer or a phone call to see what really happens. If you are in US now, I can give you a call.
For the driver issue, it is our next plan.

ArduCAM, 2018-07-10:
Contact me with admin@arducam.com if you want.

samerneergaard, 2018-07-10:
Definitely think communication would be great on this because it may affect other machines if it really is a driver issue and not just my error.

Unfortunatley, the work day has ended so I will not have access to my setup until tomorrow NYC time in the morning. As long as that works for you I will be glad to talk through it.

ArduCAM, 2018-07-10:
Send me your phone number, I will call you tomorrow morning.

samerneergaard, 2018-07-11:
After talking it through. The windows drivers worked hopefully they will be updated and changes will be made soon to the repository. Thanks for the help Lee!

closed 2018-07-11 by samerneergaard