ArduCAM/ArduCAM_USB_Camera_Shield #47: Using SDK with Python 3.3 using ctypes

Frans66, 2018-08-29:
Hi,
As there is no Python 3.x .so library available, I'm trying to connect using ctypes module.
Scan function works ok. Please tell me what's wrong in code snippet below as AutoOpen and Open both return '1' instead of '0' whatever I try.

code:
```
ctypes import *
import os

print (os.getcwd())

ArduSDK = CDLL ("/home/welcome/ArduCam/Python/libArduCamLib.so")

Scan = ArduSDK.ArduCam_scan  
Scan.restype = c_ushort

USBId = (c_ulong*100)()
VID = c_ushort(0x52cb)

res = Scan(byref(USBId), VID)
print ("result: ", res,"USB Id: ", USBId[0], USBId[1], "VID: ", VID)

ArduOpen = ArduSDK.ArduCam_open
ArduOpen.restype = c_ushort

AutoOpen = ArduSDK.ArduCam_autoopen
AutoOpen.restype = c_ushort

class CFG(Structure):
    _fields_ = [
        ("u32CameraType" , c_ulong),
        ("u32Width" , c_ulong),
        ("u32Height" , c_ulong),
        ("u32UsbVersion" , c_ulong),
        ("u8PixelBytes" , c_ulong),
        ("u16Vid" , c_ulong),
        ("u32Size" , c_ulong),
        ("u32SensorShipAddr" , c_ulong),
        ("u8PixelBits" , c_ubyte),
        ("emI2cMode" , c_uint),
        ("emImageFmtMode" , c_uint)
        ]

camhandle = c_ulong(0)

cfg = CFG()
cfg.u32Width = 2592
cfg.u32Height = 1944
cfg.u32UsbVersion = 1
cfg.u8PixelBytes = 1
cfg.u16Vid = 0x52cb
cfg.u8PixelBits = 8
cfg.u32SensorShipAddr = 0x78
cfg.emI2cMode = 2
cfg.emImageFmtMode = 2

res = ArduOpen(byref(camhandle), byref(cfg), USBId[1])
#res = AutoOpen(byref(camhandle), byref(cfg))
print ("result AutoOpen: ", res,' camhandle: ',camhandle)
```

output when run:
```
runfile('/home/welcome/ArduCam/Python/forumexample.py',` wdir='/home/welcome/ArduCam/Python')
/home/welcome/ArduCam/Python
result:  1 USB Id:  1 0 VID:  c_ushort(21195)
result AutoOpen:  1  camhandle:  c_ulong(93967354460208)

```
Thanks on beforehand for your help.

ArduCAM, 2018-08-30:
Thanks for your report. I guess you forgot to add sudo before running script. Would you please let me know your python version?
We are also trying to update our Python library for 3.x now, and will issue a update soon.

ArduCAM, 2018-08-30:
Can you send me an email to admin@arducam.com, I have an update version for python 3.x today and will send you for testing.