ArduCAM/ArduCAM_USB_Camera_Shield #57: OV5642 cfg nor json file seem to work

Frans66, 2018-09-23:
Trying to run Python3 version of USB master for windows and Linux_x86. Both show same issue.
Output from Windows version:
```
color mode 0
Serial: ÿÿÿÿ-ÿÿÿÿ-ÿÿÿÿ
Capture began, rtn_val =  0
Error capture image, rtn_val =  65317
Error capture image, rtn_val =  65314
Error capture image, rtn_val =  65314
Error capture image, rtn_val =  65314
```

Also new Windows USBTest program shows same issue. Old Windows USB Test program works flawlessly though. Can you please indicate what I'm doing wrong?

ArduCAM, 2018-09-25:
Would you please send me the hardware setup photos to admin@arducam.com ?

Frans66, 2018-09-25:
While making the pictures I noticed camera hardware after a while didn't react at all to any version of software. USB board does, but camera doesn't. I'll first order new hardware and see if that solves the issue

ArduCAM, 2018-09-26:
Please don't order any new board before sending me the your hardware setup then I will give you some suggestions.

sparky002, 2018-10-05:
I get a similar error but instead of 65314, I get 65312. Tried multiple linux computers running ubuntu 16, and ros kinetic. Tried the ros package, and python 2 package, different usb cables, a AR0134 and MT9V022 camera going through a arducam usb 2.0 breakout board. Made sure to change the config file to match the camera in the associated launch files.

Hydrorion, 2018-11-20:
Good Morning Everyone,

I am sorry but I almost have the same issues indicating the following message : "**Error capture image, rtn_val =  65317**". I've been looking for a while about it on different topics but without any conclusion. Do you guys have any leads or solution to fix it ?

I am working on ArduCAM AR0134 global shutter camera and I try to launch the device on ROS.

Thank you for your attention and have a nice day !

Sincerely,

guthubing, 2019-02-13:
Hi,
also the same problem with OV5642. Hardware below: 
[image]
Thanks!

ArduCAM, 2019-02-14:
@guthubing I can't tell the version of the USB camera shield, is it Rev.B, Rev.C or Rev.D?
To diagnose the problem, you can open the camera from the windows GUI, and try to read the OV5642 sensor register address 0x000A and 0x000B, it should read 0x56, and 0x42 respectively. 
I need you feedback to isolate this issue, before we go ahead to next level.

ArduCAM, 2019-02-14:
@guthubing you can also try other config files from: [www.arducam.com/downloads/modules/industrial/Config/OV5642](www.arducam.com/downloads/modules/industrial/Config/OV5642)

guthubing, 2019-02-14:
@ArduCAM Thanks for reply.
It is Arducam USB2.0 Camera Shield, not Arducam shield.
I tried the camera with windows GUI. The value that is read by RegAddr. 0x000A/B is 0x0. Besides, it doesn't work with the notice "USB transfer data index error".
[image]
I checked also the USB cable, connection and driver, which work well during the test.

guthubing, 2019-02-14:
@ArduCAM And by trying the python demo code ArduCam_Py_Demo.py the result looks like this:
('color mode', 0)
---------------------
Serial: --
('Capture began, rtn_val = ', 0L)
('Error capture image, rtn_val = ', 65317L)
('Error capture image, rtn_val = ', 65314L)
('Error capture image, rtn_val = ', 65314L)
('Error capture image, rtn_val = ', 65314L)
-----------------------
thanks.

ArduCAM, 2019-02-14:
@guthubing sorry I made a mistake. the ID register address is 0x300A and 0x300B, please try it again and get back to me about the result.

guthubing, 2019-02-14:
@ArduCAM It works this time.
[image]
[image]

ArduCAM, 2019-02-14:
@guthubing you can also try other config files from: [www.arducam.com/downloads/modules/industrial/Config/OV5642](www.arducam.com/downloads/modules/industrial/Config/OV5642)
especially the YUV setting?

guthubing, 2019-02-14:
@ArduCAM It doesn't work for the new config file:
C:\Users\xuexi>python C:\Users\xuexi\Dropbox\Abschlussarbeit\Arducam\ArduCAM_USB_Camera_Shield-master\ArduCAM_USB_Camera_Shield-master\Windows\Python\Streaming_demo\ArduCam_Py_Demo.py C:\Users\xuexi\Dropbox\Abschlussarbeit\Arducam\ArduCAM_USB_Camera_Shield-master\ArduCAM_USB_Camera_Shield-master\python_config\OV5642_QVGA_YUV_54fps.json
Traceback (most recent call last):
  File "C:\Users\xuexi\Dropbox\Abschlussarbeit\Arducam\ArduCAM_USB_Camera_Shield-master\ArduCAM_USB_Camera_Shield-master\Windows\Python\Streaming_demo\ArduCam_Py_Demo.py", line 221, in <module>
    if camera_initFromFile(config_file_name):
  File "C:\Users\xuexi\Dropbox\Abschlussarbeit\Arducam\ArduCAM_USB_Camera_Shield-master\ArduCAM_USB_Camera_Shield-master\Windows\Python\Streaming_demo\ArduCam_Py_Demo.py", line 53, in camera_initFromFile
    config = json.load(open(fialeName,"r"))
  File "C:\Python27\lib\json\__init__.py", line 291, in load
    **kw)
  File "C:\Python27\lib\json\__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "C:\Python27\lib\json\decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "C:\Python27\lib\json\decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
Shall I change JSON to Javascript file at first?

ArduCAM, 2019-02-14:
@guthubing  I mean you copy these config files to the Config folder of the Windows GUI tool and test on our GUI first before moving to python or other platform.
And please try the OV5642 YUV setting first, make sure if it is working.

guthubing, 2019-02-14:
@ArduCAM The OV5642 YUV could work with strange color:
[image]
But the other 3 config files can not work with warning "USB transfer data index error".

ArduCAM, 2019-02-14:
@guthubing 
it is working now, please press the button on the lower right corner to switch the display mode to get the correct color.
You still didn't tell me the board revision of your USB2 camera board.

guthubing, 2019-02-14:
@ArduCAM Thanks!. It is USB 2.0 shield Rev.B.

ArduCAM, 2019-02-14:
Change the display mode, is it working correctly?
We will check the work around for Rev.B hardware for JPEG config files soon.

guthubing, 2019-02-14:
@ArduCAM It works generally well, with warning "bad frame recieved".
Besides, how could I expand the resolution from 320*240 to original 2592*1944 of OV5642?
thanks.

ArduCAM, 2019-02-14:
@guthubing  I added another full 5MP YUV config OV5642_YUV_1944.cfg in the link: [www.arducam.com/downloads/modules/industrial/Config/OV5642](www.arducam.com/downloads/modules/industrial/Config/OV5642)
You can try that .

guthubing, 2019-02-14:
@ArduCAM Unfortunatly it doesn't work with YUV_1944 file. After clicking open button, there's only still image from last capturing with warning "bad frame recieved" appearing 2~3 times per second. When the camera is closed and restarted again, there's warning "USB device create error", no matter what config file it is now. So I have to reconnect the USB again.
[image]

ArduCAM, 2019-02-15:
@guthubing please be patient, we will work that today and give you update.

ArduCAM, 2019-02-15:
@guthubing I am still working on the maximum resolution config file.
In the mean time, would you please help me check if the following config files work, they only work for REV.B board.
[www.arducam.com/downloads/modules/industrial/Config/OV5642/OV5642_QVGA_JPEG_REVB.cfg](www.arducam.com/downloads/modules/industrial/Config/OV5642/OV5642_QVGA_JPEG_REVB.cfg)
[www.arducam.com/downloads/modules/industrial/Config/OV5642/OV5642_JPEG_720p_REVB.cfg](www.arducam.com/downloads/modules/industrial/Config/OV5642/OV5642_JPEG_720p_REVB.cfg)
[www.arducam.com/downloads/modules/industrial/Config/OV5642/OV5642_JPEG_1080p_REVB.cfg](www.arducam.com/downloads/modules/industrial/Config/OV5642/OV5642_JPEG_1080p_REVB.cfg)

ArduCAM, 2019-02-15:
@guthubing 
I found there is still some bugs on Rev.B board.
The above confg file, you have to open->close and open again then click play to get the video out.

ArduCAM, 2019-02-15:
@guthubing 
The following config for 5MP, but you have to load the smaller resolution config file and open then close and load this config:
[www.arducam.com/downloads/modules/industrial/Config/OV5642/OV5642_JPEG_QSXGA_TEST.cfg](www.arducam.com/downloads/modules/industrial/Config/OV5642/OV5642_JPEG_QSXGA_TEST.cfg)
The Rev.B board is quite old, now we have update the board  to REV.D.

guthubing, 2019-02-15:
@ArduCAM 
OV5642_QVGA_JPEG_REVB.cfg could work for only few seconds, they it stopped with warning "USB transfer time out error".
OV5642_JPEG_720p_REVB.cfg worked well.
For OV5642_JPEG_1080p_REVB.cfg there's image, but unrecognizable. Some part of the image break into pieces and come out randomly.

guthubing, 2019-02-15:
@ArduCAM OV5642_JPEG_QSXGA_TEST.cfg did work, but with about 1 fps.

guthubing, 2019-02-21:
@ArduCAM Could you send me the link of OV5642 full resolution config file formal edition once you finish it? Thanks!

amalhub, 2019-07-17:
Hi @guthubing / @ArduCAM 
I'm having the same problem. Can you please share the config files with me, please? The given links are not working anymore to download the cfg files. Specially looking for (OV5642_JPEG_720p_REVB.cfg)