ArduCAM/ArduCAM_USB_Camera_Shield #59: RaspberryPi + MT9J001

nifunk, 2018-09-25:
Dear all,

I have been trying to get the raspberry Pi demo working with the MT9J001 module but I always encounter the error:
rtn_val =  65316
I have also tried both, running with the RaspberryPi demo files as well as with the ROS integration. Do You have any advice on how this issue could be resolved?

Thanks a lot in advance

ArduCAM, 2018-09-26:
can you send me the screen shot?

nifunk, 2018-09-26:
I unfortunately do not have the screenshot right now, but i am using the usb 3.0 usb camera shield together with an raspberry pi 3b+ with the setup shown in the pictures,...
[image]
[image]
[image]

erikbeall, 2018-11-09:
I have the same error and can provide the terminal output. Sometimes a few images come across, usually none do. I run a "rostopic hz /arducam/camera/image_raw" in another terminal, which gives the following output:
`
~/catkin_ws/src/arducam_usb2_ros/src $ rostopic hz /arducam/camera/image_raw
subscribed to [/arducam/camera/image_raw]
no new messages
no new messages
no new messages
average rate: 0.653
        min: 1.531s max: 1.531s std dev: 0.00000s window: 2
average rate: 0.678
        min: 1.419s max: 1.531s std dev: 0.05611s window: 3
no new messages
`

And in the other terminal I run the ros node, like so:
`
~/catkin_ws/src/arducam_usb2_ros/src $ python /home/ubuntu/catkin_ws/src/arducam_usb2_ros/src/arducam_ros_node.py _config_file:="/home/ubuntu/catkin_ws/src/arducam_usb2_ros/camera_config_files/AR0134_960p_Mono.json" _frame_id:="vis_cam"
color mode 0
USB VERSION: 2
Serial: AU2S-1834-0022
Capture began, rtn_val =  0
USB_CAMERA_DATA_LEN_ERROR       RECEIVE_LENGTH:1234058
Error capture image, rtn_val =  65316
USB_CAMERA_DATA_LEN_ERROR       RECEIVE_LENGTH:1233920
Error capture image, rtn_val =  65316
USB_CAMERA_DATA_LEN_ERROR       RECEIVE_LENGTH:1233920
Error capture image, rtn_val =  65316
USB_CAMERA_DATA_LEN_ERROR       RECEIVE_LENGTH:1233920
Error capture image, rtn_val =  65316
USB_CAMERA_DATA_LEN_ERROR       RECEIVE_LENGTH:1233920
Error capture image, rtn_val =  65316
USB_CAMERA_DATA_LEN_ERROR       RECEIVE_LENGTH:1233920
Error capture image, rtn_val =  65316
<truncated, note they're not always the same>
`

Additional debugging information: RPi 3, ArduCam shield with AR0134 and either color or mono json file, have updated to most recent git on Oct 29th (commit 23ef4d68db7852697d536bc0de58f8ad44b16e0b), have tried changing delays in ROS code, stripping down, etc, to no avail. Nothing else is running except a roscore, on ros-kinetic in Ubuntu ubiquity, kernel 4.9.41-v7+. I can get more images when running the plain Raspberry Pi python2 demo. For the ROS code, I've copied the Raspberry Pi sdk .so over to the ROS area before running. Any ideas?

ehong-tl, 2018-11-10:
Hi @nifunk ,
Can you give me the Vendor ID and Device ID for USB 3 shield? Cause I don't have USB 3 shield, so the ROS node I created is only meant for USB 2 shield currently.

Hi @erikbeall 
I will try it on my RPi 3 next week and come back to you later.

ArduCAM, 2018-11-11:
@erikbeall 
Please note that the AR0134 is configured with 1280x964 resolution, so the 1233920 is the correct size, so please modify your resolution value to 1280x964.

ehong-tl, 2018-11-12:
hi @erikbeall 

From my testing with MT9N001 arducam usb 2 shield (9 MP camera), it seems like RPI 3 is not fast enough to run the ROS node, especially with high resolution camera.

iMhack, 2019-10-04:
Hi,
Do you think that the USB3 of Raspberry Pi 4B will be fast enough ?

iMhack, 2019-11-25:
I answer myself. I got a RPi4 and I can read the MT9J001 with a pimped .cfg to record 1080p. More test later.
(I still got a lot of 65316 messages. The images are laced badly when I move my mouse/keyboard).