ArduCAM/ArduCAM_USB_Camera_Shield #60: Controlling 2 AR0134 Cameras at Once

atfields43, 2018-09-25:
Hi, I would like to control 2 AR0134 cameras at the same time, using a raspberry pi. I have a USB camera shield for each. When I use the scan function, it returns that there are 2 cameras, and that the addresses are [1,2]. While I can open, capture, and close the first camera, When I try to open the second camera, I get a USB_CREATE_ERROR.

`camnum = ArducamSDK.Py_ArduCam_scan()
ret,handle,rtn_cfg = ArducamSDK.Py_ArduCam_open(cfg,camnum[1][0])
ret2,handle2,rtn_cfg2 = ArducamSDK.Py_ArduCam_open(cfg,camnum[1][1])`

and while ret = 0 indicating a valid open 
ret2 =  0xFF01 indicating a USB_CREATE_ERROR

The one thing is I do not include a parameter for the scan function, which says it includes a ArduCamIndexInfo parameter, because I do not see anywhere in the documentation what exactly to put for that. 

Any help would be appreciated.

ehong-tl, 2018-09-26:
Hi @atfields43 

What I've found out is that the camera index number return by the scan function from python SDK is not correct.
The true index number of the camera always start at 0 and increase in order (0,1,2,3,...) based on the position of the camera in serial number list output by scan function. You can see how I open multiple cameras based on serial number of the camera in ROS python node from line 114 to 129. By the way, please allow some delay (time.sleep(...)) after scan to prevent any error.

ArduCAM, 2018-09-26:
@ehong-tl we will work on this issue soon.

atfields43, 2018-09-26:
Still getting an error when i use [0,1] instead of [1,2], and I can't use the rospy script because it gives an error about import ArducamSDK. I can work around if I just know what serial_selection looks like? I have no idea what rospy.get_param("~camera_serial") returns