ArduCAM/ArduCAM_USB_Camera_Shield #61: Color Readout of MT9F002 orange

icfaust, 2018-10-02:
Hello,

I'm currently using a MT9F002 14MP with the lens which comes with a MT9J001. Even though I think it doesn't have an IR cut filter, my images (jpg from the streaming_demo) are coming out orange using the MT9F002 .cfg file.  Is there a guide to the register and/or camera parameters so that I can try to color correct the jpg output? Is there something fundamental I am not doing properly? Note: I haven't looked at the raw data yet either (pre bayer).

ArduCAM, 2018-10-02:
We one shipped MT9F001 camera right now, so please use MT9F001 config file and try it again.

icfaust, 2018-10-02:
Ah.  I am using a transplanted MT9F002 from another non-Arducam camera board.  I can send you an email about this project (I bought several MT9J001 boards for this purpose).

Using the MT9F001 cfg the images are coming out rather white/black and aren't showing the proper colors (maybe a gain or saturation issue?)

ArduCAM, 2018-10-02:
Please send your hardware setup photo to admin@arducam.com