ArduCAM/ArduCAM_USB_Camera_Shield #66: Raspberry pi External Trigger

habib-Boloorchi, 2018-10-08:
I want to manipulate a camera to give me image 20 fps by external trigger but it made for 2 cameras and I just want to use one camera and an Imu with different timestamps.  would you tell me which parameters have to be changed to do this job?
Would you tell me how to change it to make a Rosbag from it?

Razbotics, 2019-08-03:
Exactly same thing i want to do. Did you achieve it? I will buy the camera if external triggering works.

habib-Boloorchi, 2019-08-03:
The trigger works but the data wont come to raspberry consistantly. It has too much delay and delays are change. It is not stable. I am keep searching to find a better choice.

Razbotics, 2019-08-03:
Did u tried with your laptop or pc?

habib-Boloorchi, 2019-08-03:
I will try it next week, and call u back.

Razbotics, 2019-08-05:
Yes please do it and let me know. So that i can buy the camera module