ArduCAM/ArduCAM_USB_Camera_Shield #67: External trigger example gives error when trigger pin is pulled high

ehong-tl, 2018-10-10:
Hi,

I'm using MT9N001 camera with USB2 shield board.

I've trouble capturing image using external trigger image capture example in new SDK.

I keep getting alternating error as follows each time when I pulled the trigger pin high.

_Take picture.
('Take picture fail, ret_val = ', 65318)
Take picture.
('Take picture fail, ret_val = ', 65316)
Take picture.
('Take picture fail, ret_val = ', 65318)
Take picture.
('Take picture fail, ret_val = ', 65316)
Take picture.
('Take picture fail, ret_val = ', 65318)
Take picture.
('Take picture fail, ret_val = ', 65316)_

May I know what's going on here, and how to fix it?

ArduCAM, 2018-10-10:
Would you please let me know your hardware setup?  How do you send the trigger signals? Please send to support@arducam.com

ehong-tl, 2018-10-10:
Hi,

I didn't change much in terms of hardware setup.
I send the trigger signal by just connecting Vcc pin to Trigger pin with jumper wire to pull it high.
[image]

ArduCAM, 2018-10-10:
It is saft to trigger this signal with pulse from Arduino like GPIO, the trigger input default should be logic low, and pull logic high for 10uS or so to trigger the capture.

ehong-tl, 2018-10-10:
Hi, I've tried to send 10us pulse using Arduino to the camera, but there is no response, it 'doesn't even print out _Take picture._ line.

ArduCAM, 2018-10-11:
I realized that your camera only has 8MByte frame buffer, it can't hold 9MP image sensor. We will build 32MByte frame buffer version later.

ehong-tl, 2018-10-11:
So the current USB2 shield with 8 MB buffer does not support external trigger image capture?

CONFIDENTIALITY: This email is intended solely for the person(s) named and may be confidential and/or privileged. If you are not the intended recipient, please delete it, notify us and do not copy, use, or disclose its contents.
Towards a sustainable earth: Print only when necessary. Thank you.

ArduCAM, 2018-10-11:
No, I think it supports external trigger, but the MT9N001 9MP camera has exceeded the maximum frame buffer of current camera shield.

ehong-tl, 2018-10-15:
Hi @habib-Boloorchi ,

I connect one of Arduino's digital pin to Trigger pin on the camera board (not the USB shield) and Arduino's ground pin to camera board's ground pin.