ArduCAM/ArduCAM_USB_Camera_Shield #74: AR0134 at 50 fps with  arduCAM USB2 shield

hefortunato, 2018-10-20:
I tested the arduCAM AR0134 using an arduCAM USB2 shield on Ubuntu/python2 using the demo  and everything runs OK giving color, 1280x964 resolution at ~15 fps.

(I used the config file : "ArduCAM_USB_Camera_Shield-master/Linux_x86/Python2/JSON_Config_Files/AR0134_960p_Color.json")

I want to get color, 640x480 resolution at ~50 fps. I understand from the documentation that I can do this using binning, line/column skip or setting the frame size to 640x480pixels but I could not find the right combination of register values.

Can you say me if that is possible (640x480 at the 50fps with the USB2 shield) and give me the correct register values combinations?

Thanks in advance.

ArduCAM, 2018-10-21:
Please try to add the following settings to your config file
REG= 0x3032, 0x0022
REG= 0x306E, 0x9000
REG =0x3030, 0x006C
The size setting should be:
SIZE      = 640,484 
because there are addtional 4 lines for embedded data

hefortunato, 2018-10-22:
Thanks a lot!, 
it worked fine: color 640x480 at 47 fps.
I had also to reset bit 7 of 0x30B0 to obtain color images.

ArduCAM, 2018-10-22:
Can you share your modified json file and cloes this issue?

hefortunato, 2018-10-22:
Follows my modified json file:
[AR0134_960p_Color_binned_640_484_OK.zip](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/files/2501643/AR0134_960p_Color_binned_640_484_OK.zip)

closed 2018-10-22 by hefortunato