ArduCAM/ArduCAM_USB_Camera_Shield #76: ROS does not work

habib-Boloorchi, 2018-10-23:
When I run ros code in both raspberry pi (strech, kinetic ros) or in ubuntu14.04(indigo) I get same error. this is the error:

habib@Habib:~/catkin_ws/src/arducam_usb2_ros/launch$ roslaunch arducam_node.launch 
... logging to /home/habib/.ros/log/fc66a430-d710-11e8-9a69-bc77370b7345/roslaunch-Habib-4297.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/__init__.py", line 307, in main
    p.start()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 746, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 718, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 652, in _recurse_load
    default_machine, is_core, verbose)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 654, in _recurse_load
    n = self._node_tag(tag, context, ros_config, default_machine, verbose=verbose)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 406, in _node_tag
    self._param_tag(t, param_ns, ros_config, force_local=True, verbose=verbose)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 256, in _param_tag
    vals = self.opt_attrs(tag, context, ('value', 'textfile', 'binfile', 'command'))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 202, in opt_attrs
    return [self.resolve_args(tag_value(tag,a), context) for a in attrs]
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 316, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 329, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 142, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 188, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: arducam_usb2_ros
what I changed is just in launch's configure changed it to MT9v034_VGA.json 
would you help me with this issue?

habib-Boloorchi, 2018-10-24:
hey @ehong-tl thank you for your help 
I reinstalled ros again and got this error:
gging to /home/pi/.ros/log/7122dfea-d799-11e8-8678-b827eb0229c3/roslaunch-raspberrypi-10790.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:39779/

SUMMARY
========

CLEAR PARAMETERS
 * /cam0/

PARAMETERS
 * /cam0/arducam_trigger_ros_node/camera_serial: 
 * /cam0/arducam_trigger_ros_node/config_file: /home/pi/catkin_w...
 * /cam0/arducam_trigger_ros_node/frame_id: cam0
 * /cam0/arducam_trigger_ros_node/horizontal_flip: False
 * /cam0/arducam_trigger_ros_node/vertical_flip: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /cam0/
    arducam_trigger_ros_node (arducam_usb2_ros/arducam_trigger_ros_node.py)

ROS_MASTER_URI=http://localhost:11311

process[cam0/arducam_trigger_ros_node-1]: started with pid [10809]
Traceback (most recent call last):
  File "/home/pi/catkin_ws/src/arducam_usb2_ros/src/arducam_trigger_ros_node.py", line 7, in <module>
    from ImageConvert import *
  File "/home/pi/catkin_ws/src/arducam_usb2_ros/src/ImageConvert.py", line 2, in <module>
    import ArducamSDK
ImportError: /home/pi/catkin_ws/src/arducam_usb2_ros/src/ArducamSDK.so: wrong ELF class: ELFCLASS64
[cam0/arducam_trigger_ros_node-1] process has died [pid 10809, exit code 1, cmd /home/pi/catkin_ws/src/arducam_usb2_ros/src/arducam_trigger_ros_node.py __name:=arducam_trigger_ros_node __log:=/home/pi/.ros/log/7122dfea-d799-11e8-8678-b827eb0229c3/cam0-arducam_trigger_ros_node-1.log].
log file: /home/pi/.ros/log/7122dfea-d799-11e8-8678-b827eb0229c3/cam0-arducam_trigger_ros_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
pi@raspberrypi:~/catkin_ws/src/arducam_usb2_ros/launch $ 
and I dont need any convert I need monochrome raw images for two cameras with triggers.
would you help me this issue

ehong-tl, 2018-10-24:
Hi @habib-Boloorchi,

The ArducamSDK.so file in the ROS folder is meant for PC, not ARM processor like Raspberry Pi.

Please replace the ArducamSDK.so in the arducam_usb2_ros/src/ folder with the one in RaspberryPi/Python/Streaming_demo/, and try to run the node again.

ehong-tl, 2018-10-24:
> thank you @ehong-tl<https://github.com/ehong-tl> that one fixed and again this is the new error:

Did you run setup.sh during installation before running the node?

ERROR: cannot launch node of type [arducam_usb2_ros/arducam_trigger_ros_node.py]: can't locate node [arducam_trigger_ros_node.py] in package [arducam_usb2_ros]
No processes to monitor

CONFIDENTIALITY: This email is intended solely for the person(s) named and may be confidential and/or privileged. If you are not the intended recipient, please delete it, notify us and do not copy, use, or disclose its contents.
Towards a sustainable earth: Print only when necessary. Thank you.

habib-Boloorchi, 2018-10-24:
@ehong-tl thank you I did that and now I have this error:
process[cam0/arducam_trigger_ros_node-1]: started with pid [15949]
Traceback (most recent call last):
  File "/home/pi/catkin_ws/src/arducam_usb2_ros/src/arducam_trigger_ros_node.py", line 11, in <module>
    from cv_bridge import CvBridge, CvBridgeError
ImportError: No module named cv_bridge
[cam0/arducam_trigger_ros_node-1] process has died [pid 15949, exit code 1, cmd /home/pi/catkin_ws/src/arducam_usb2_ros/src/arducam_trigger_ros_node.py __name:=arducam_trigger_ros_node __log:=/home/pi/.ros/log/46eb3c94-d79f-11e8-b0c9-b827eb0229c3/cam0-arducam_trigger_ros_node-1.log].
log file: /home/pi/.ros/log/46eb3c94-d79f-11e8-b0c9-b827eb0229c3/cam0-arducam_trigger_ros_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

ehong-tl, 2018-10-24:
I think there’s something wrong with your ROS, it should come with cv_bridge module by default.

habib-Boloorchi, 2018-10-26:
hey @ehong-tl , I wanted to be thankful for your help I fixed it and it works well. but I have a question do you know how to change the trigger code to get data from two cameras by a trigger. And do you know if I can get grayscale from cameras, converting to grayscale is time-consuming and I cannot sync it with my IMU properly.

ehong-tl, 2018-10-26:
Hi @habib-Boloorchi ,

You can run two trigger nodes simultaneously with different namespace and camera serial number, once you trigger both the camera externally, it will publish to two different topic based on the namespace.

But if you want both image topic with same timestamp that sync with your IMU, you have to create your own node to subscribe to these 2 image topic and your IMU, and use ApproximateTime synchronizer to republish the image topic with synced timestamp.

For getting grayscale image directly from camera, you have to play with your camera config file register value.

habib-Boloorchi, 2018-11-22:
Hi @ehong-tl ,
do you now why I get this error in this image and why I have those + line in my image and how to remove them:
[image]

ehong-tl, 2018-11-24:
Hi @habib-Boloorchi ,

I believe RPi is too slow to run the current script. You can try run separate thread for image capture and image reading.

vipergts, 2019-05-29:
> Hi @habib-Boloorchi,
> 
> The ArducamSDK.so file in the ROS folder is meant for PC, not ARM processor like Raspberry Pi.
> 
> Please replace the ArducamSDK.so in the arducam_usb2_ros/src/ folder with the one in RaspberryPi/Python/Streaming_demo/, and try to run the node again.

I solve the problem 
with this comment 
Thank you!