ArduCAM/ArduCAM_USB_Camera_Shield #78: Error when capturing with demo code for python ubuntu 18.04

amchagas, 2018-10-24:
now that I'm able to run the example code, I get the following error:

('Error capture image, rtn_val = ', 65318)

Any ideas or suggestions on what this might be?

thanks!

UCTRONICS, 2018-10-26:
@amchagas 
Don't worry and our engineer will help you solve the problem.
Does it keep going wrong or occasionally go wrong? Can you see any images?
Please attach me some pictures for your error messages.

amchagas, 2018-10-26:
Hi, thanks for the reply,
here is a screenshot of the message (note that I tried running the python script as regular user and as su). 

Also I get the error all the time, and can't see any images.

[image]

UCTRONICS, 2018-10-26:
@amchagas 
('open fail,rtn_val = ', 65281) means can't find the USB camera, you can run the 'lsusb' command to check if the camera connects normally. if it connects normally, you will see the ID 52cb:52cb or ID 04b4:03f1.
What's more, in order to solve your problem quickly, please tell me the UC number of your bottom board.

amchagas, 2018-10-26:
Hey,

So lsusb gives me 52cb:52cb and the board I have is UC-391 rev.b

Thanks!

UCTRONICS, 2018-10-26:
@amchagas 
Does it work normally now? If not, please add sudo to enable superuser permissions.

amchagas, 2018-10-26:
no, unfortunately not... 
I'm adding sudo, but this is what I get:
[image]

amchagas, 2018-10-26:
One question that might be silly, is the current code compatible with all arducam boards? (v1, v2 etc)

UCTRONICS, 2018-10-26:
@amchagas 
Yes, this code is compatible with all of the Arducam boards.
Maybe the camera has not connected normally.

amchagas, 2018-10-26:
Hum. You mean the cable from the camera to the arducam board?

UCTRONICS, 2018-10-26:
@amchagas 
Yes, If reconnect it and still not work normally, you can try it on windows platform.
For the detail operations, please refer to the link https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Windows/GUI/USBTest

amchagas, 2018-10-26:
Great, so changing the cable made everything work! Thanks for the support!

Also the cable was flipped on the camera side, (the way it came in the box.
Just a heads up, so that other customers don't have the same problem.. )

UCTRONICS, 2018-10-26:
@amchagas 
Great to hear you have solved your problem. 
Thanks for your advice.

closed 2019-05-25 by amchagas