ArduCAM/ArduCAM_USB_Camera_Shield #83: ask for creation WVGA monochrome mt9v034 json config

habib-Boloorchi, 2018-11-23:
Hi, 
I have seen that everyone uses mt9v034 with this resolution: 750 * 480 in monochrome. I need this for my project. would you pls create this configuration. do I need a need a new sdk for it? I am using it for my raspberry pi in ROS and I am not kind of new to hardware and configuration. would you give me some instruction about its confiig file too?
Thank you,

ArduCAM, 2018-11-27:
Would you please let me know what is the exact problem you have right now？

habib-Boloorchi, 2018-11-27:
I saw in the data sheet that we can get 750*480 monochrom(grayscale) image from mt9v034. But when I change the format to 4 or change the size to 750*480 .), it does not show any thing in rqt_image_show.
I use this camera with raspberry pi and get data by trogger node of ros.

ArduCAM, 2018-11-28:
Can you run the default setting without any modification?
Simply modify the config file might not be able to work properly.

habib-Boloorchi, 2018-11-29:
yes I can run default one it is the image of the results andits rate:
[image]
I want to get grayscale data because I am using this image for specific project. can you help me with this?