ArduCAM/ArduCAM_USB_Camera_Shield #84: chessboard noise in MT9J001

hyperOP, 2018-12-06:
Hi, I have noticed strange noise on images from MT9J001 camera. At first it seemed random to me, but then I've noticed that it has regular, chessboard scheme.  I have made simple test and took several images of white surface, choose small part of the image (10x10 pixels, randomly chosen) and calculated average of every pixel for set of images. It seems that there is about 30 intensity levels difference between "white" and "black" fields.  In the specification of the sensor there is mentioned "Bayer monochrome pattern" but I haven't found  anything about it in the web.  Is it physical grayscale filter? Whats the purpose of this?

[image]

ArduCAM, 2018-12-07:
It is really monochrome, but due to the architecture of the sensor, it should be compatible its color version. So there are 4 registers to control the 2x2 pixel gains separately which is used for color sensor R, GR,GB,B gains.

hyperOP, 2018-12-07:
Is it possible to change it in the SDK or do you mean "manually" with my own code? Can you tell me gain values for those pixels?

ArduCAM, 2018-12-11:
You can change the register through our SDK API function call.
www.arducam.com/downloads/modules/industrial/MT9J003_RR_A.pdf Page 11, 
Change the following registers for different gains
REG = 0x0206, 0x24 
REG = 0x0208, 0x2a 
REG = 0x020a, 0x2a
REG = 0x020c, 0x24

lionfish0, 2019-05-18:
Glad I found this issue. Below is part of an image with the 'chessboard' effect that @hyperOP mentioned. I assume this is the same thing.
[image]

Is there a reason these registers aren't correct out-the-box? Also - besides pointing it at a blank white background and working it out from there, is there a more principled way of getting this scaling correct?

ArduCAM, 2019-05-18:
Please find the explanations from: http://www.arducam.com/docs/camera-breakout-board/10mp-mt9j001-mt9j003/

lionfish0, 2019-05-18:
Thanks,
I tried adding                
```
                ["0x0206","0x0024"],
                ["0x0208","0x002a"],
                ["0x020a","0x002a"],
                ["0x020c","0x0024"]
```
to the [AR0134_960p_Mono.json](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/blob/master/python_config/AR0134_960p_Mono.json) file inside the `"register_parameter" `list, but the registers didn't appear to update (I read them back and they didn't change). The image didn't change either.

Maybe I'm missing something or misunderstood...?

lionfish0, 2019-05-18:
ah, I found on a link from the page you linked to that I maybe should try setting

```
Reg    | Description
-------|------------
0x3056 | green1_gain
0x3058 | blue_gain
0x305A | red_gain
0x305C | green2_gain
0x305E | global_gain
```
I'll try editing those instead...

ArduCAM, 2019-05-20:
Congratulations!