ArduCAM/ArduCAM_USB_Camera_Shield #88: Raspberry Pi external trigger "firmware version not supported"

PoojaMangla89, 2019-01-16:
Hi, I am using Raspberry PI to run external trigger demo with MT9J001 and OV5640 cameras. I am using Arducam version UC-391 Rev.B. 

When I run the external trigger demos with both, I get an error "USB_BOARD_FW_VERSION_NOT_SUPPORT_ERROR" on setting "external trigger" mode. I think it means that the firmware version is not supported.  

However when I set the mode as "CONTINUOUS" and run the streaming demo, it runs successfully and I am able to stream the video. 

Has anyone faced this and knows about a potential solution to this?

smartexpt, 2019-02-08:
I'm also having this problem Using UC-391 Rev.B with AR0134 Camera. Any solution?

ArduCAM, 2019-02-17:
The Rev.B firmware doesn't support trigger mode, you can only get a frame our of the stream video.
You can find our lastest Rev.D board for advanced feature.