ArduCAM/ArduCAM_USB_Camera_Shield #94: Unable to initialize camera sensor from linux library

joncppl, 2019-02-08:
I'm using the USB3 variant (rev. c) with MIPI adapter board connected to an OV5647 (rpi "spy cam")

Running the demo program `sudo ArduCam_Demo  ../../../cpp_config/OV5647_MIPI_RAW10_edit.yml`  yields:
```
Serial: AU3S-1903-0010
capture thread create successfully.
read thread create successfully.
Capture began, rtn_val = 0
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
```

(From the header I can tell that 65318 == 0xFF26 == USB_CAMERA_USB_TIMEOUT_ERROR)

However, I tried the Windows USBTest GUI/MFC application with `OV5647_MIPI_RAW10.cfg`, and it worked.

Strangely, when I booted back into linux, `ArduCam_Demo` now works with the same configuration file.

I thought my problem was solved until I power cycled the ArduCam. It reverted to no longer works in Linux.

I repeated the process, and determined that running the USBTest application once in windows consistently enables it to work on my linux setup. 

Btw, I have repeated this cycle on up to date Arch/Manjaro, Ubuntu 18.04, and Ubuntu 16.04. All exhibit the same behaviour. 

Any advice? 

(Also this is impossible for me to debug since the source for libArduCamLib.so is, as far as I can tell unavailable, a practice that ought to be admonished)

ArduCAM, 2019-02-17:
Sorry for my late reply. It seems like the configration file for linux is not correct, we will verify the config file next week then get back to you.

ArduCAM, 2019-02-18:
Please check the update config file and SDK library.

joncppl, 2019-02-19:
Thank you for your response.

Unfortunately, the update did not solve my issue.
I pulled the latest git repo, rebuilt the demo application, and still get the same error message.

```
[jonathan@jonathan-leopard Streaming_demo]$ git reset --hard
HEAD is now at 92d854c add more sensor config files
[jonathan@jonathan-leopard Streaming_demo]$ git clean -fd
Removing images/
[jonathan@jonathan-leopard Streaming_demo]$ git pull
Already up to date.
[jonathan@jonathan-leopard Streaming_demo]$ rm ArduCam_Demo
[jonathan@jonathan-leopard Streaming_demo]$ g++ -L. -I. -std=gnu++11 -g ArduCam_Demo.cpp ./Arducam_SDK/libArduCamLib.a -lusb-1.0 -lpthread -lpthread `PKG_CONFIG_PATH=/opt/opencv3/lib/pkgconfig/ pkg-config --cflags --libs opencv` -o ArduCam_Demo
[jonathan@jonathan-leopard Streaming_demo]$ sudo LD_LIBRARY_PATH=/opt/opencv3/lib/ ./ArduCam_Demo ../../../cpp_config/OV5647_MIPI_RAW10.yml
device num:1
index:   0      Serial:AU3S-1903-0010
Serial: AU3S-1903-0010
capture thread create successfully.
read thread create successfully.
Capture began, rtn_val = 0
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
```

I tried the updated library in Windows, it still works,
```
C:\Users\Jonathan\Desktop\ArduCAM_USB_Camera_Shield-master\Windows\Cpp\Streaming_demo\x64\Debug>ArduCam_test.exe ..\..\..\..\..\cpp_config\OV5647_MIPI_RAW10.yml
device num:1
index:   0      Serial:AU3S-1903-0010
Serial: AU3S-1903-0010
capture thread create successfully.
read thread create successfully.
Capture began, rtn_val = 0
-------------------10 fps
-------------------29 fps
-------------------27 fps
-------------------11 fps
-------------------18 fps
-------------------30 fps
-------------------28 fps
-------------------29 fps
-------------------29 fps
-------------------28 fps
-------------------23 fps
-------------------26 fps
-------------------28 fps
Read thread stopped
Capture thread stopped.

Press ENTER to exit...
```

I rebooted into Linux, it worked once. I unplugged/plugged back in the ArduCam, then I get error 65318.

joncppl, 2019-03-27:
Any new information?

UCTRONICS, 2019-03-27:
@joncppl 
Please try this configuration files
[OV5647.zip](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/files/3011249/OV5647.zip)

joncppl, 2019-03-27:
Thank you for trying to help me work through this issue.

I tried both configuration files. Neither succeeded in initialising the sensor in Linux.
Both worked in Windows, however.

I believe the problem is that the sensor is not receiving the I2C commands when initialised by the Linux library.

For example, If I use `Arducam_test.exe OV5647_MIPI_RAW8_usb2.yml` from windows, it works in 8 bit mode.

If I switch over to **Linux**, and then use `./ArduCam_Demo OV5647_MIPI_RAW8_usb2.yml`, it works. But if I use 10bit mode `./ArduCam_Demo OV5647_MIPI_RAW10.yml`, I get the error
`USB_CAMERA_DATA_LEN_ERROR	RECEIVE_LENGTH:2510928`

Hence I think the sensor never received the I2C commands setting it to 10bit mode.

joncppl, 2019-03-27:
I will try the 8bit mode configuration with the USB2 revision of the ArduCam and see if I can get a framerate acceptable for my application, since the USB2 revision is working fine with Linux.

UCTRONICS, 2019-03-28:
@joncppl 
[OV5647_MIPI_RAW10.zip](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/files/3016384/OV5647_MIPI_RAW10.zip)

UCTRONICS, 2019-03-28:
@joncppl 
Do you connect the USB2.0 or USB3.0 interface?

joncppl, 2019-03-28:
I'm afraid I am seeing the same result with the most recent configuration.

I am connecting the ArduCam to a USB3.0 port (not a hub). The OS detects it as "SuperSpeed Gen 1 USB".

In the demo code cameraCfg.usbType is `USB_3`, so "board_parameter_dev3_inf3" is loaded, if that helps.

UCTRONICS, 2019-03-29:
@joncppl 
This problem is very strange, we in the start interface test is no problem, only on the USB2.0 interface bandwidth problems, I wrote a simple detection function to test the register write and a few small changes have been made to the configuration file (I'm not sure whether he will be able to solve your problem), using make ArduCam_Demo_test compiled test program, using methods and ArduCam_Demo is the same, this is my operation: 
[image]
[Streaming_demo.zip](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/files/3021379/Streaming_demo.zip)

joncppl, 2019-03-29:
Interesting results

```
[jonathan@maho Streaming_demo]$ sudo LD_LIBRARY_PATH=$(pwd)/Arducam_SDK ./ArduCam_Demo ./OV5647_MIPI_RAW10.yml
device num:1
index:   0	Serial:AU3S-1903-0010
Serial: AU3S-1903-0010
capture thread create successfully.
read thread create successfully.
Capture began, rtn_val = 0
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
^C
[jonathan@maho Streaming_demo]$ sudo LD_LIBRARY_PATH=$(pwd)/Arducam_SDK ./ArduCam_Demo_test ./OV5647_MIPI_RAW10.yml
device num:1
index:   0	Serial:AU3S-1903-0010
error on register 0x103,current value = 0x00
error on register 0x100,current value = 0x01
error on register 0x3036,current value = 0x52
error on register 0x3821,current value = 0x03
error on register 0x5002,current value = 0x41
error on register 0x380e,current value = 0x05
error on register 0x3503,current value = 0x00
error on register 0x350b,current value = 0xf8
error on register 0x3501,current value = 0x7b
error on register 0x3502,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x350b,current value = 0xf8
error on register 0x350b,current value = 0xf8
error on register 0x3501,current value = 0x7b
error on register 0x3502,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x350b,current value = 0xf8
error on register 0x3501,current value = 0x7b
error on register 0x3502,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x350b,current value = 0xf8
error on register 0x3501,current value = 0x7b
error on register 0x3502,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x350b,current value = 0xf8
error on register 0x3501,current value = 0x7b
error on register 0x3502,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x350b,current value = 0xf8
error on register 0x3501,current value = 0x7b
error on register 0x3502,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x350b,current value = 0xf8
error on register 0x3501,current value = 0x7b
error on register 0x3502,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x3212,current value = 0x00
error on register 0x350b,current value = 0xf8
Serial: AU3S-1903-0010
capture thread create successfully.
read thread create successfully.
Capture began, rtn_val = 0
-------------------29 fps
-------------------29 fps
-------------------30 fps
```
`ArduCam_Demo_test` works!

I believe the cause of the change is that `configBoard` is called after `writeSensorRegs` in the test code.

UCTRONICS, 2019-04-01:
@joncppl 
This is a very strange phenomenon. If so, Windows should have the same problem with the instance program.  And it does not have this problem on my ubuntu16.0. I will continue to trace this problem.