ArduCAM/ArduCAM_USB_Camera_Shield #96: Error in Capture Image on Raspberry Pi

GhostSystemV2, 2019-04-10:
Hi!
I'm having problems with the AR0134+USB3.0 Rev.C shield, when i run the Streaming_demo in a Raspberry Pi, some frames are corrupted or the terminal says "Error capture image. rtn_val =" 6531L, and i can't exit the capturing process so i have to kill the script.

In windows the camera works without problems.

[image]

glddiv, 2019-04-11:
@GhostSystemV2 
Try to reduce the value of the 0x3030 register.
[image]

GhostSystemV2, 2019-04-11:
Thanks! i have reduced the value, but it looks that it has also a relation with the Frecuency of the input, in the datasheet it talks about the PLL multiplier, but why, it is defined in the register_parameter_dev_3 ?

glddiv, 2019-04-16:
@GhostSystemV2 
The 0x3030 register has a default value. On USB 3.0 we want to get faster and we need to increase its value.
[image]

GhostSystemV2, 2019-04-17:
Okay, perfect, thanks for the information.

closed 2019-04-17 by GhostSystemV2