ArduCAM/ArduCAM_USB_Camera_Shield #98: 'USB transfer data lack error'

hyoooo2, 2019-05-01:
I tried 'usb2 camera shield' and OV5642 camera.
Installation usb driver on windows 8.1 64bits was successfully done.
Open device '0  AU2S-1848-0018' / 'Load OV5642_JPEG_QSXGA.cfg' from usbtest.exe had no problem and it showed 'USB camera init success!' in a message box.
But Click play button was failed with error message 'USB transfer data lack error'.
SDK guide has no further explanation about this error message.
I tried 32bit usbtest.exe. Same result.
What is this error? How can i fix it?

ArduCAM, 2019-05-01:
Would you please show me the hardware setup photo? Which kind of board version of your USB camera shield? For more info , please read: http://www.arducam.com/docs/camera-breakout-board/5mp-ov5642/software-guide/

hyoooo2, 2019-05-01:
Here is version number of board i bought.
UV-391
Rev.D

ArduCAM, 2019-05-01:
I uploaded several config files in the guide : http://www.arducam.com/docs/camera-breakout-board/5mp-ov5642/software-guide/  please try it again.
Thanks.

hyoooo2, 2019-05-01:
Thanks for quick response and configuration files.
All of three configuration are working though 'USB transfer data index error' just after 'play'.
But frame rate is terribly limited to 2-3 fps under only 'Display' option.
I can hardly think my laptop(i7-3720qm, 16GB ram) has no strength to properly handle this streaming data. Any suggestion/help will be appreciated.

ArduCAM, 2019-05-01:
The RAW or YUV data is too big for USB2.0 port at full resolution. I am optimize the frame rate now, will keep update soon. Please be patient.

ArduCAM, 2019-05-01:
Using OV5642_RAW_2592x1944.cfg , add extra line REG = 0x3011, 0x18 or manually set from the GUI, you can get 8fps@5MP