ArduCAM/ArduCAM_USB_Camera_Shield #101: Issue with ArduCam_isFrameReady returning 0 on AR0134

MoBaT, 2019-05-16:
Hey!

I just recently purchased an AR0134 and am trying to run the test projects on your repo. When I run the example Arducam_Global_Shutter using AR0134_960p_Mono.yml, I get:

```
device num:1
index:   0	Serial:AU2S-1917-0008
Found 1 devices.
Serial: AU2S-1917-0008
```

But the line `ArduCam_isFrameReady(tempHandle)` returns 0 all the time. I used a AC/DC meter to make sure I'm getting power to the board and am successfully getting power.

What I have tried doing:

1. I have tried running the other example External_trigger_demo and I get the same issue.
2. I have tried to ArduCam_setMode() to CONTINUOUS_MODE and that did not work either.
3. I have tried the code with the USE_SOFT_TRIGGER turned on and off to no avail.

What do you think is the issue of what I cannot get any data from the camera?

Thanks!

vipergts, 2019-05-27:
I have same problem 

i typed 
`sudo ./ArduCam_Global_Shutter_Demo ../../../cpp_config/AR0134_960p_Color.yml `

and got the message 

`device num:1`
`index:   0	Serial:AU2S-1917-0009`
`Found 1 devices.`
`Serial: AU2S-1917-0009 `

And Nothing happen 

How can i do to see the demo using Ar0134?

glddiv, 2019-05-29:
@MoBaT @vipergts 
Is Streaming_demo working properly?

vipergts, 2019-05-29:
No.

glddiv, 2019-05-29:
@vipergts 
What is the error message?
Can you tell me your hardware information?

MoBaT, 2019-05-29:
@glddiv I am running the camera on Inspiron 7559. I have tried to run it on LINUX and WINDOWS but both of them did not work for me. I have also tried using the GUI app for windows that you guys have developed and that did not work either.

glddiv, 2019-05-30:
@MoBaT 
Please tell me what error message is output from the software, and can you take a photo of the USB board and camera? Or tell me the exact model, similar to UC-xxx, Rev.X, I need this information to locate the problem.

vipergts, 2019-05-30:
> @vipergts
> What is the error message?
> Can you tell me your hardware information?

-> There is no error message just not working 

device num:1
index:   0	Serial:AU2S-1917-0009
Found 1 devices.
Serial: AU2S-1917-0009
(This is the end nothing happen )

and my hardware is AR0134 +USB Camera shield (UC391REV.D)

and i check AR0134 in windows USB Test 
It works well only AR0134_960p.cfg  
that means Hardware is no problem 

and i compare "AR0134_960p.cfg" and "AR0134_960p_Color_yml" 
there is some different things in Timing_settings (COARSE_INTEGRATION_TIME)

And i also try in ROS but not working 
This is the Error message 

color mode 2
USB VERSION: 2
Serial: AU2S-1917-0009
Capture began, rtn_val =  0
USB_CAMERA_DATA_LEN_ERROR	RECEIVE_LENGTH:1235188
Error capture image, rtn_val =  65316

sometimes i`ve got the message 

 Error capture image, rtn_val =  65318

glddiv, 2019-05-30:
@vipergts You seem to be using the External_trigger_demo or Arducam_Global_Shutter example. The Streaming_demo example does not generate an error without outputting an error message. Try re-plug the hardware and running the Streaming_demo example.
The ROS example seems to be normal. If you want to use the ROS example to display an image, you need a receiver to display it.

frascuelillo, 2019-05-30:
> -> There is no error message just not working
>
> device num:1
> index: 0	Serial:AU2S-1917-0009
> Found 1 devices.
> Serial: AU2S-1917-0009
> (This is the end nothing happen )
>
> and my hardware is AR0134 +USB [...]

Have you tried with USBTest.exe  ?, at least you can verify that it is not a hardware problem.

vipergts, 2019-05-31:
> Have you tried with USBTest.exe ?, at least you can verify that it is not a hardware problem.

Yes, i tried with USBTest.exe in windows 
and it works well only AR0134_960p.cfg

vipergts, 2019-05-31:
> @vipergts You seem to be using the External_trigger_demo or Arducam_Global_Shutter example. The Streaming_demo example does not generate an error without outputting an error message. Try re-plug the [...]

I`ve tried 
1. RassberryPi->Cpp->Arducam_Global_shutter 
2. ROS/arducam_usb2_ros
3. Windows->USBTest->Realease->USBTest.exe 

but Only works in 3. USBTest.exe 
And i try Streaming_demo like you said 
it works very well 

Also i want to use it in ROS 
What is the receiver ?

glddiv, 2019-05-31:
@vipergts 
1. Why is Arducam_Global_shutter not showing?
The Arducam_Global_shutter example uses the external trigger mode, you need to send a high level to the USB Board's Trigger pin to trigger the camera to take a picture. [https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/blob/master/Data/Arducam_UC-391_Rev_D.png](url)
2. How to receive and display ROS images.
We just uploaded an example of ROS receiving and displaying. You can update the new file (arducam_usb2_ros/src/display_test.py) to your folder and execute the catkin_make command. After running the roslaunch arducam_usb2_ros arducam_node.launch command you can use the rosrun arducam_usb2_ros display_test.py command to display the image.
3. How to change the exposure parameters.
Modifying the exposure parameters requires modifying the register of the camera sensor. Take the sensor AR0134 as an example. The exposure register address of AR0134 is 0x3012. You can modify it in the configuration file, or modify it in the code:
C++: ArduCam_writeSensorReg(cameraHandle, 0x3012, the value you want to write);
Python: ArducamSDK.Py_ArduCam_writeSensorReg(handle, 0x3012, the value you want to write)

lionfish0, 2019-07-09:
Hi,
I'm having a similar problem. The Streaming Demo is working fine, and I've found I can make it take photos just when the trigger pin is high, by setting 0x3064 to 0x1982. So I might be able to use that, although it takes an arbitrary number of photos & it doesn't quite work very neatly (seems to take between 0 and 2 photos if I just use a brief trigger pulse of ~50ms).
Anyway, I found that the external trigger demo doesn't work for the reasons given above, the ArduCam_isFrameReady is always zero. Interestingly the 'flash trigger' pin on the camera goes high, which maybe suggests the camera /is/ taking a photo?

ArduCAM, 2019-07-09:
@lionfish0 Can you show me your hardware setup? And we also update the config files for the AR0134/AR0135 external trigger, can you try that?
Thanks.

lionfish0, 2019-07-09:
Hi, thanks. OK: I've tried the new config file AR0135_1280x964_ext_trigger_M.json.
It's a bit complicated* to show, but I've checked that the trigger pin goes high, and I can confirm that the flash pin goes high in response (just checked again, with a multimeter and with an LED - my scope's not available at the moment).

Also the trigger pin does seem to make it take photos + it works in the non-trigger version.

*There's a separate board I've made to select which of four flashes are used a system for monitoring the battery voltage and a sensor to check if the flash has fired.

lionfish0, 2019-07-09:
OK, I've tried reading from 0x303C (which is frame-status, standby-status and framesync), using this line of code in the while loop in the external trigger example:
print(ArducamSDK.Py_ArduCam_readSensorReg(handle,0x303C))
The function returns an error code and the value of the register. The error code is always zero. I also read frame count, which I don't really understand so much,
each pair of numbers below is 16bit hex for frame-count, frame-status
00FF, 0002
00FF, 0002
00FF, 0002
<-- trigger sent about here
FFFF, 0000 
0000, 0000
00FF, 0002
00FF, 0002
00FF, 0002

Does that help?

lionfish0, 2019-07-09:
OK - I've just tried something and it might be working, but I don't know for sure & I definitely don't understand why.

I added the line:
        ArducamSDK.Py_ArduCam_softTrigger(handle)
at the start of the while loop - I was expecting it to take photos, but nothing happened until I triggered the external trigger... why I'd need the soft trigger AND the external trigger, I don't know. Anyway, it took a photo when externally triggered.
Any idea why this works?

For clarity about the line I added:
```
    while running and len(handles):
        ArducamSDK.Py_ArduCam_softTrigger(handle)
        for i in range(len(handles)):
            handle = handles[i]
            value = ArducamSDK.Py_ArduCam_isFrameReady(handle)
            if value == 1:
                getAndDisplaySingleFrame(handle,i)
        cv2.waitKey(10)
```

lionfish0, 2019-07-09:
I forgot to say I'm using a mono AR0135, maybe that's a difference?

glddiv, 2019-07-10:
@lionfish0 
Hi, lionfish0
Call ArduCam_setMode(cameraHandles[i], EXTERNAL_TRIGGER_MODE); the function will cause the USB Board to work in the external trigger mode, you need to send the trigger signal from the USB Board's Trigger pin. Can you try it?

lionfish0, 2019-07-10:
> @lionfish0
> Hi, lionfish0
> Call ArduCam_setMode(cameraHandles[i], EXTERNAL_TRIGGER_MODE); the function will cause the USB Board to work in the external trigger mode, you need to send the trigger signal from the USB Board's Trigger pin. Can you try it?

Hi. The line of code is the same as is used in the code I'm using (the ArduCam_Ext_Trigger_Demo.py file). I'm less clear what you mean by "you need to send the trigger signal from the USB Board's Trigger pin". I'm currently sending the trigger signal _to_ the trigger pin on the _ArduCam_. By "USB board" do you mean the USB shield? (I can't see a trigger pin on there) also I'm not sure what you mean by sending it "from" that board.
In summary: I think the hardware side is probably ok as it almost all works (the flash is triggered, the status registers change, and if I set up the softTrigger then it does take a photo), so it's maybe a software issue. I don't know if the AR0135 is different from the 134, or if I've messed something else up, or maybe an issue with the build on the raspberry pi, or something... but anyway. It's working for me at the moment with the work around with the softTrigger line added, so I'll leave it - unless it breaks again.

glddiv, 2019-07-11:
@lionfish0 
Sorry, the translation may have misunderstood what I meant.
Let's clarify some issues.
> Also the trigger pin does seem to make it take photos + it works in the non-trigger version.

> I'm less clear what you mean by "you need to send the trigger signal from the USB Board's Trigger pin". I'm currently sending the trigger signal to the trigger pin on the ArduCam.

Is this trigger pin on the camera or on the USB camera shield?
[image]
Is it left or right?

lionfish0, 2019-07-11:
Ah! I am so sorry! Yeah, I was using the trigger on the camera. I didn't even spot the trigger on the USB camera shield. It works now. Fantastic. It's the UC-391 Rev D. Ugh, sorry - I didn't realise that it wasn't the camera trigger pin - what is that for? Anyway - thanks again so much for your help!

glddiv, 2019-07-19:
@lionfish0 
Sorry for your bad experience, I think this is caused by the incomplete publication of our information, and we will improve it.

lionfish0, 2019-07-19:
No worries - It was probably caused by me not looking carefully enough! Thanks for being so responsive and helpful, really appreciate it.

pritorius, 2019-08-03:
> @vipergts
> 
> 1. Why is Arducam_Global_shutter not showing?
>    The Arducam_Global_shutter example uses the external trigger mode, you need to send rising edge signal to the USB Board's Trigger pin to trigger the camera to take a picture.
> 
> [image]
> 
> 1. How to receive and display ROS images.
>    We just uploaded an example of ROS receiving and displaying. You can update the new file (arducam_usb2_ros/src/display_test.py) to your folder and execute the catkin_make command. After running the `roslaunch arducam_usb2_ros arducam_node.launch` command you can use the `rosrun arducam_usb2_ros display_test.py` command to display the image.
>    Note:If you want to learn more, you need to read the [official ROS tutorial.](http://wiki.ros.org/)
> 2. How to change the exposure parameters.
>    Modifying the exposure parameters requires modifying the register of the camera sensor. Take the sensor AR0134 as an example. The exposure register address of AR0134 is 0x3012. You can modify it in the configuration file, or modify it in the code:
>    C++: `ArduCam_writeSensorReg(cameraHandle, 0x3012, the value you want to write);`
>    Python: `ArducamSDK.Py_ArduCam_writeSensorReg(handle, 0x3012, the value you want to write)`

> @vipergts
> 
> 1. Why is Arducam_Global_shutter not showing?
>    The Arducam_Global_shutter example uses the external trigger mode, you need to send rising edge signal to the USB Board's Trigger pin to trigger the camera to take a picture.
> 
> [image]
> 
> 1. How to receive and display ROS images.
>    We just uploaded an example of ROS receiving and displaying. You can update the new file (arducam_usb2_ros/src/display_test.py) to your folder and execute the catkin_make command. After running the `roslaunch arducam_usb2_ros arducam_node.launch` command you can use the `rosrun arducam_usb2_ros display_test.py` command to display the image.
>    Note:If you want to learn more, you need to read the [official ROS tutorial.](http://wiki.ros.org/)
> 2. How to change the exposure parameters.
>    Modifying the exposure parameters requires modifying the register of the camera sensor. Take the sensor AR0134 as an example. The exposure register address of AR0134 is 0x3012. You can modify it in the configuration file, or modify it in the code:
>    C++: `ArduCam_writeSensorReg(cameraHandle, 0x3012, the value you want to write);`
>    Python: `ArducamSDK.Py_ArduCam_writeSensorReg(handle, 0x3012, the value you want to write)`

Do I need to someting ( i.e. a button) to create an exernal trigger? Or the code is itself going to send a rising edge signal ?

glddiv, 2019-08-05:
@pritorius 

> Do I need to someting ( i.e. a button) to create an exernal trigger?

Yes!

> Or the code is itself going to send a rising edge signal ?

You can use software triggering if you want (provided that the "USB camera shield" is in external trigger mode and the camera is not in the external trigger mode because the software trigger does not send a signal to the camera)

f5-zahmed, 2019-08-05:
@glddiv 
> You can use software triggering if you want (provided that the "USB camera shield" is in external trigger mode and the camera is not in the external trigger mode because the software trigger does not send a signal to the camera)

Thanks for the reply. I was able to make software triggering work. 
But what I actually need is this: 
1. There will be 3 cameras. The middle one will be the MASTER. The left and right camera will be SLAVE.
2. The 2 SLAVES have wired connection from the MASTER cameras External trigger interface to their external trigger input via 2 wires.
3. When I send a software trigger to MASTER, the SLAVES will also be triggered via the wired connection. 
4. After triggering is done, I will collect frame from all 3 cameras.

Is the above doable?

glddiv, 2019-08-06:
@f5-zahmed 
> 1. There will be 3 cameras. The middle one will be the MASTER. The left and right camera will be SLAVE.
> 2. The 2 SLAVES have wired connection from the MASTER cameras External trigger interface to [...]

There is a problem here, the software trigger only sends the trigger signal to the "USB camera shield", so SLAVES can't receive any trigger signal.

pritorius, 2019-08-06:
Thanks for the reply. So if I connect the external trigger interface pin to the 2 slaves trigger pin, I will not be able to trigger the slaves? 
Thanks again for the reply.

f5-zahmed, 2019-08-06:
@glddiv 
> I think you need a microcontroller or something else to send a trigger, such as an arduino.

I am going to create a pulse signal from arduino to trigger all 3 cameras together.

glddiv, 2019-08-07:
@pritorius 
> If I setup the 3 cameras ( middle camera is MASTER and Left and right Cameras are SLAVE. And the 2 SLAVES external trigger pin is connected to the trigger interface of MASTER via wire) as the picture above, and send a software trigger to only the MASTER, will the SLAVES be triggered?

SLAVES will not be triggered, only MASTER will be triggered.

lionfish0, 2019-08-29:
Has this behaviour changed in newer versions?
I just bought two more ar0135s (and USB shields).
They're not triggering with the pin on the shield any more, but still trigger on the one on the camera itself... It seems like the shield is the bit that has changed as I've tried with the original camera I was using.

Maybe something else is going on. Is there anything I should be doing to the shield to make this work?

BMaxBrooks, 2019-11-13:
Hello,

I appear to be having an identical problem as lionfish0 commented, any solution?

Thank you,
Brian

BMaxBrooks, 2019-11-13:
Hello,

I appear to be having an identical problem as lionfish0 commented, any solution?

Thank you

tuananh8693, 2021-07-14:
Dear all,
I want to use an external trigger with Arducam Cmos AR0134
However, when using Arducam SDK on Pi 4, FPS is 10fps
Can I increase the speed to at least 25fps?
Please help me