ArduCAM/ArduCAM_USB_Camera_Shield #102: USB Timeout Transfer Error Using AR0135C

smoise, 2019-05-21:
Hello!

I'm having problems with the AR0135 C+USB3.0 Rev.C shield on Windows. when I run the USBTEST.exe, I'm able to connect perfectly fine and it says USB camera init success, but once I hit play I get the USB Transfer Timeout Error. Same thing happens over on the Python side and I get error 65318.

Looking at some of the other issues, I was thinking it could be because the configuration files for the AR0134 is not compatible with AR0135 C. I'm currently using the AR1034 config because there is no AR0135 C settings for python or C.

[image]
[image]

Any help would be much appreciated.

ArduCAM, 2019-05-22:
AR0134 is almost the same as AR0135, you can find more info to diagnose the issue from: http://www.arducam.com/docs/camera-breakout-board/1-2mp-ar0134-ar0135-global-shutter/software-guide/ 
Please check the camera ID first to make sure the camera is well connected.

sammo2828, 2019-06-22:
This is exactly the same problem I'm having.
Following @ArduCAM 's link, here is some new information.
[image]
Expecting RegAddr Read 0x0000 to be 0x754 as in the screenshot above, but I am getting 0x4155, which I guess is an error code.
I have quite a few pairs of Arducam UC-425 Rev.C USB3 shields with AR0135-C cameras. So far 3 out of 3 have the same error and I don't want to open the packaging of the remaining.
Also, I noticed that only 1 of the 3 will light up the LED on the shield board located to the left under the UC-425 Rev.C printed on the board.
Please advise.

ArduCAM, 2019-06-24:
If you can't read the correct register value, it means there is something wrong with the I2C interface between the USB camera shield to the sensor board.
Since you have so many cameras with the same problem I guess there is something wrong with the USB3 camera shield. Can you swap to another USB3 camera shield?

sammo2828, 2019-06-24:
All of the USB3 camera shields have the same result. Maybe I am not connecting the ribbon cable properly? Do you have video/images showing a correct connection?

ArduCAM, 2019-06-25:
@sammo2828 
Please refer to this image I attached for you
[image]
[image]

sammo2828, 2019-06-25:
Thanks. I think the problem was that I loaded the AR0134 cfg before loading the AR0135 cfg. After restarting the GUI and loading the AR0135 cfg it seems to play video.

Still, do you have any ideas why only one of my shields has a LED light turned on?

> I noticed that only 1 of the 3 shields will light up the LED located to the left under the UC-425 Rev.C printed on the board.

ArduCAM, 2019-06-25:
I'm not very sure about the LED issue. Are all the cameras and USB camera shields working now?

sammo2828, 2019-06-25:
Yes, working now. No more timeout transfer error. Thanks!