ArduCAM/ArduCAM_USB_Camera_Shield #105: USB_CAMERA_DATA_LEN_ERROR

ldg810, 2019-06-05:
Error log. The error occurs randomly. (Jeton Nano - AR0134)

```
Error capture image, rtn_val =  65316
Error capture image, rtn_val =  65316
USB_CAMERA_DATA_LEN_ERROR	RECEIVE_LENGTH:971776
Error capture image, rtn_val =  65316
USB_CAMERA_DATA_LEN_ERROR	RECEIVE_LENGTH:185344
Error capture image, rtn_val =  65316
USB_CAMERA_DATA_LEN_ERROR	RECEIVE_LENGTH:971776
Error capture image, rtn_val =  65316
```

ArduCAM, 2019-06-05:
Please let me know the USB camera shield board version.

ldg810, 2019-06-05:
I am using usb 2.0 shield(http://m.vctec.co.kr/product/detail.html?product_no=11927&cate_no=65&display_group=1)

ArduCAM, 2019-06-06:
Thanks for your info, I guess it is old image.
The Rev.A and Rev.B has no frame buffer, so the performance is not good, you have to reduce the pixel clock speed by adjust the PLL register settings.
For Rev.C and Rev.D camera there is 32Mbyte frame buffer, there is no such issue.

stmoon, 2019-06-06:
How do I adjust the PLL register settings?

ArduCAM, 2019-06-06:
You can try to reduce the pll_multiplier  resgister 0x3030 with lower value.

ldg810, 2019-06-07:
I checked my usb camera shield. The Rev was D.

However, I tried to reduce the pixel clock speed as follows. It worked without error. (AR0134_960p_Color.json, "0x0018" => "0x0016" in third line of register_parameter)

```
"register_parameter" : [	
		["0x3028","0x0010"],
		["0x302e","0x0001"],
		["0x3030","0x0018"],   ==> ["0x3030","0x0016"],
```

Could you check the situtation? I guess I do not need to change parameter because my usb shield is Rev. D

ArduCAM, 2019-06-09:
You have to make sure if the USB2 camera shield is connected to the USB root hub, and no other USB device connected to the same root hub, or else they are rush for the USB bandwidth.

MarianCrisan, 2023-01-24:
Hi, 

I purchase a 108MP arducam kit, which came with Usb3 camera shield version UC-593 Rev. C.

The linux platform I use, is: Linux 3.18.71-perf #1 SMP PREEMPT Thu Oct 13 17:57:05 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux

I tried first with the config file ArduCAM_USB_Camera_Shield/Config/USB3.0_UC-593 Rev.C/MIPI/ArduCAM_108MP/ArduCAM_108MP_MIPI_2Lane_RAW8_4000x3000_12fps.cfg and I got the error below:

device num:1
index:   0	Serial:AU3S-2244-0007
Serial: AU3S-2244-0007
capture thread create successfully.
read thread create successfully.
Waiting for the camera 10secondes
readImage_thread running ...
Capture began, rtn_val = 0
USB_CAMERA_DATA_LEN_ERROR	RECEIVE_LENGTH:3616696
Error capture image, rtn_val = 65316
USB_CAMERA_DATA_LEN_ERROR	RECEIVE_LENGTH:3616688
Error capture image, rtn_val = 65316
USB_CAMERA_DATA_LEN_ERROR	RECEIVE_LENGTH:7233448
Error capture image, rtn_val = 65316
USB_CAMERA_DATA_LEN_ERROR	RECEIVE_LENGTH:3616696
Error capture image, rtn_val = 65316
USB_CAMERA_DATA_LEN_ERROR	RECEIVE_LENGTH:3616688
Error capture image, rtn_val = 65316

Can you please advice how to correct this?