ArduCAM/ArduCAM_USB_Camera_Shield #106: Problem with open ArduCam usb 2.0 - MT9J001

pbieguszewski, 2019-06-06:
I have problem with open the ArduCam board. When I call 'ArduCam_open' this function return me 'USB_CAMERA_USB_CREATE_ERROR' flag. If I call this function several time then I can use the board. In your sample program(\Windows\Cpp\Streaming_demo\ArduCam_test) also exist problem.

Board init:

```
ArduCamCfg cameraCfg = {
  {}, // camera type
  {}, // vendor id
  3664, // width
  2748, // height
  1, // pixel size
  8, // bit depth
  0x20, // I2C address
  {}, // size ??
  {}, // usb type
  I2C_MODE_16_16, // I2C mode
  FORMAT_MODE_MON, // die mode
  64 // TRANS_LVL ??
};
```

Enviroment: Windows 10 x64, compiler VS2017

ArduCAM, 2019-06-09:
Sorry for the trouble, would you please send me a video of this problem to support@arducam.com
thanks.

pbieguszewski, 2019-07-11:
Ok, problem fixed. I forgot add 'sleep(2s)' before I call 'ArduCam open' function.

closed 2019-07-11 by pbieguszewski