ArduCAM/ArduCAM_USB_Camera_Shield #107: MT9V034 initialization failed

vvostri, 2019-06-07:
Hi,

I just got a MT9V034 board that I am using with USB 2.0 shield under Raspbian. The configuration file is https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/blob/master/python_config/MT9V034_VGA.json

The camera fails to initialize with error code 65281.

lsusb command with just the USB shield correctly returns ID 52cb:52cb, but this device disappears when I connect MT9V034 to the USB shield.

Same camera shield works fine with the AR0134 camera module.

Is this the faulty MT9V034 board, or am I missing something?

Thank you.

ArduCAM, 2019-06-09:
Please provide your hardware setup photos and software error message screenshot?

vvostri, 2019-06-09:
Raspberry Pi 3B+
USB2 Camera Shield
MT9V034
[image]

`sudo python ArduCam_Py_Demo.py MT9V034_VGA.json 0`
**('open fail, rtn_val = ', 65281L)**
`lsusb`
No camera detected
[image]

vvostri, 2019-06-09:
Turns out the cable was not all the way in, so now the camera board is detected, but still not initializing properly:

Setup: Raspberry Pi 3b+, USB2 camera shield, MT9V034
[image]

Output:
[image]

Two initialization files I tried:
https://raw.githubusercontent.com/ArduCAM/ArduCAM_USB_Camera_Shield/master/python_config/MT9V034_VGA.json
`{
	"camera_parameter" : {
		"CFG_MODE":1,
		"TYPE":"MT9V034",
		"SIZE":[640,480],
		"BIT_WIDTH":8,
		"FORMAT":[0,0],
		"I2C_MODE":1,
		"I2C_ADDR":"0x90",
		"TRANS_LVL":64
	},
	"board_parameter" : [
	],
	"board_parameter_dev2" : [	
		["0xd7","0x4600","0x0100","0x01",["0x00"]],
		["0xd7","0x4600","0x0100","0x01",["0x05"]]
	],
	"board_parameter_dev3_inf2" : [	
		["0xf3","0x0000","0x0000","0x00",[]],
		["0xf9","0x0000","0x0000","0x00",[]]
	],
	"board_parameter_dev3_inf3" : [	
		["0xf3","0x0000","0x0000","0x00",[]],
		["0xf9","0x0000","0x0000","0x00",[]]
	],
	"register_parameter" : [	
		["0x0003","0x01e0"],
		["0x0004","0x0280"]
	],
	"register_parameter_dev3_inf2" : [	
		["0x0003","0x01e0"],
		["0x0004","0x0280"]
	],
	"register_parameter_dev3_inf3" : [	
		["0x0003","0x01e0"],
		["0x0004","0x0280"]
	]
}`

http://arducam.com/downloads/modules/industrial/MT9V022/MT9V034_VGA_8bit.cfg
`{
	"camera_parameter" : {
		"CFG_MODE":1,
		"TYPE":"MT9V034",
		"SIZE":[640, 480],
		"BIT_WIDTH":8,
		"FORMAT":[0,0],
		"I2C_MODE":1,
		"I2C_ADDR":"0x90",
		"TRANS_LVL":64
	},
	"board_parameter" : [
	],
	"board_parameter_dev2" : [	
		["0xd7","0x4600","0x0100","1",["0x00"]],
		["0xd7","0x4600","0x0100","1",["0x05"]],
		["0xd7","0x4600","0x0200","1",["0x00"]]
	],
	"board_parameter_dev3_inf2" : [	
		["0xf3","0x0000","0x0000","0x00",[]],
		["0xf9","0x0000","0x0000","0x00",[]]
	],
	"board_parameter_dev3_inf3" : [	
		["0xf3","0x0000","0x0000","0x00",[]],
		["0xf9","0x0000","0x0000","0x00",[]]
	],
	"register_parameter" : [
		["0x0003","0x01e0"],
		["0x0004","0x0280"],
		["0x000D","0x0320"],
		["0x00af","0x0000"],
		["0x0020","0x03c7"],
		["0x0024","0x001b"],
		["0x002b","0x0003"],
		["0x002f","0x0003"]
	],
	"register_parameter_dev3_inf2" : [	
		["0x0003","0x01e0"],
		["0x0004","0x0280"]
	],
	"register_parameter_dev3_inf3" : [	
		["0x0003","0x01e0"],
		["0x0004","0x0280"]
	]
}`

vvostri, 2019-06-10:
My mistake: I used the flex cable instead of the primary interface.

http://blog.uctronics.com/downloads/modules/industrial/ArduCAM_USB_Camera_MT9V034AN.pdf

closed 2019-06-10 by vvostri