ArduCAM/ArduCAM_USB_Camera_Shield #108: 12-bit MT9F001 Capture Returns 65318

dcargol, 2019-06-17:
Hi appologies in advance if I'm missing something obvious, but I'm kind of stumped. I'm tring to figure out how the get 12-bit data from a MT9F001 sensor using the USB3 shield.

Looking at various 8 and 12-bit configuration files for the closely related MT9J003 and MT9J001 camera sensors, it looks like all you would need to change is BIT_WIDTH from 8 to 12 and everything else gets handled within the python demo script.

However when I switch it and try to capture anything I get a "Error capture image, rtn_val =  65318" (SDK lists it as USB_CAMERA_DATA_LEN_ERROR). I've tried slowing things down by tinkering with the various ppl_multipliers and clk_dividers but it dosen't seem to make a difference beyond how quickly the errors are getting generated.

Everything works great when BIT_WIDTH is 8 but borks at 10 or 12. Any ideas? Thanks in advance!!

ArduCAM, 2019-06-17:
If you wanto get 12bit data from MT9F001 sensor using USB3 camera shield, you need another adapter board for that: https://www.uctronics.com/index.php/catalogsearch/result/?q=Arducam+Parallel+Camera+Adapter+Board+for+USB3+Camera+Shield 
Because the FPC connector on the USB3 camera shield is only 8bit, you need to feed the 12bit through the black 16bit camera  interface  see more from: http://www.arducam.com/downloads/shields/USB_Shield/USB3/ArduCAM_USB3_Camera_Shield_DS.pdf
For 16bit mode, you also need to change to this line: VRCMD = 0xF9, 0x0001, 0x0000, 0 to the config file

dcargol, 2019-06-17:
Ah that makes perfect sense, thank you!!

closed 2019-06-17 by dcargol

zachjacobs, 2020-03-17:
@KermitTheHog can you share the exact config you used here?  I am not able to get this working based on their response.