ArduCAM/ArduCAM_USB_Camera_Shield #109: Large delay when using Python bindings with MT9F001

timrae, 2019-06-19:
Steps to reproduce:

1. Connect MT9F001
1. `cd Linux_x86/Python/Streaming_demo`
1. `sudo python ArduCam_Py_Demo.py ../../../python_config/MT9F001_14MP.json`
1. Get a consistent frame rate of ~2 frames/second
1. But when I put my hand in front of the camera, I have to wait for about 3 seconds until I see the camera update

Note that in the C++ version, this delay issue does not occur, so I suspect there is an issue somewhere with the python bindings. By disabling `cv2.imshow()` in the python script, I can get up to 5 frames per second, but still the delay issue is there.

ArduCAM, 2019-06-20:
Yes,  the python version has lower speed and long latency than C/C++ examples.

timrae, 2019-06-20:
The latency is too long, can you improve, or open source the python bindings?