ArduCAM/ArduCAM_USB_Camera_Shield #110: AR0135 Error Capturing Image and USB Data Length Error

jjpaulson, 2019-06-20:
Hello,

I just got an AR0135 camera sensor with a wide angle lens to use in a visual inertial odometry system, and I am getting what I think are some sort of buffer issues when capturing images. I am using the ROS driver for the camera (in ROS melodic on Ubuntu 18.04) along with the USB 3.0 Camera Shield connected to a confirmed USB 3.0 port. I am using the config file for the AR0135 posted on its software guide, just modified into json format. I am able to get a good stream of video for a little bit, but after an extended period of time I end up getting these 'Error capture image, rtn_val = 65316' and 'USB_CAMERA_DATA_LEN_ERROR' errors along with a messed up video. Furthermore, doing anything that tries to display or use the image data, like image_view or running the visual odometry software I am trying to use, seems to almost immediately cause these errors to occur, which seems very weird to me since they should just be subscribing to the data being published to that topic from the camera.

I have tried decreasing the PLL value in the register as suggested in another issue, but that did not seem to have any effect. I also tested the camera in Windows, and it does seem to be better, but I do still get 'bad frame received' errors when the camera is running for more than a few minutes. Any help would be much appreciated. I have attached a few pictures of the errors in the terminal and what the affected images look like.

[image]
[image]
[image]

glddiv, 2019-06-20:
Hello @jjpaulson, please try this configuration file, it works fine on my ubuntu16.04. If it doesn't work, try reducing the value of register 0x3030 in the "register_parameter_dev3_inf2" block and the "register_parameter_dev3_inf2" block.
[AR0135_1280x964_8bit.zip](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/files/3308507/AR0135_1280x964_8bit.zip)

jjpaulson, 2019-06-20:
Thank you for the help. I tried your config file, and needed to add the TRANS_LVL parameter to it, but that did not fix either of the errors. I then reduced the value of register x3030 in the "register_parameter_dev3_inf2" and "register_parameter_dev3_inf3" blocks by quite a bit and this did help quite a bit. The camera works for some time and I have not seen the USB_CAMERA_DATA_LEN error at all, but after some time, I still get the "Error capture image, rtn_val = 65318" that ends up crashing the camera video. Should I try dropping the 0x3030 values even lower? I brought them down to 0x000e and 0x0025 in the inf2 and inf3 blocks respectively from their original values.

glddiv, 2019-06-21:
@jjpaulson  Can you try another computer?
I can't reproduce this problem on my computer.

jjpaulson, 2019-06-21:
@glddiv Sorry, I do not have another Linux machine around with all of the ROS stuff set up. I did, however, get around my problems by converting the provided C++ code (that was working on its own) into a ROS node itself. This seems to be working fine. I don't know what's happening with the provided python code. Thanks for your help though.

ArduCAM, 2019-06-24:
python code is not as efficent as the C++ code. The problem looks like the capture process is not fast enough to get the data before it is flushed with the incoming data.

jjpaulson, 2019-06-24:
@ArduCAM Yeah that is what I am assuming as well. The C++ Ros code has been much more reliable in preventing these errors and has been able to run continuously for hours. However, I am getting the occasional issue of several "Error capture image" errors followed by the capture crashing. Then, when I try to run the same code again, the capture can begin successfully, but it then crashes immediately. It only starts working again after unplugging the camera and plugging it back in. I am wondering if this problem is more of a USB issue than the software, since this problem showed up with the python code as well.
[image]
[image]