ArduCAM/ArduCAM_USB_Camera_Shield #112: Arducam OV5642 camera with USB shield not working

amalhub, 2019-07-17:
Hi, 
I have purchased Arducam OV5642 camera (UC-234 Rev.B) and I'm trying to connect it to my Ubuntu 16.04 pc using the Arducam USB camera shield (UC-391 Rev.D). I have tried with both Python and Cpp but nothing works. The camera initializes but doesn't return any image data. 

Below screenshot is the Cpp error. 
[image]

Below screenshot is the Python error.
[image]

Can you please help me with resolving this issue?

glddiv, 2019-07-17:
Hi @amalhub 
Can you try this configuration file?
[OV5642_Config_File.zip](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/files/3400370/OV5642_Config_File.zip)

amalhub, 2019-07-18:
Hi @glddiv 
Thanks for the configuration files. I tried the configuration files but still, get the same error. :( 
Do you have any other solutions or insight what cause these error and why the device doesn't work out of the box?

glddiv, 2019-07-18:
> Hi @glddiv
Thanks for the configuration files. I tried the configuration files but still, get the same error. :(
Do you have any other solutions or insight what cause these error and why the device doesn't work out of the box?

I am not sure what caused this error, I used the same hardware as you and it works. Can you send me the error message you printed using the USBTest program on Windows?

amalhub, 2019-07-19:
Hi @glddiv 
I was initially testing your configs in the Linux environment with Python scripts and it gave me the same error. It looks same as the screenshots in my previous comment. However when you mention that you tested the scripts on Windows environment then I did the same later. It works in Windows environment with .cfg file. Then I tested the .json file in Linux Cpp environment and got it to work as well. For some reason, the config files are not compatible with Python. However, I'm satisfied since they are working with Linux Cpp.

amalhub, 2019-07-19:
I have one more question. 
The default configurations in the camera are set to return 2592x1944 resolution with 8fps. 
I need the camera to work with a much lower resolution with a higher frame rate. Ex: 1280x720 resolution with 30fps.
Can you please show me how can I modify the configurations to achieve this?

glddiv, 2019-07-19:
> For some reason, the config files are not compatible with Python. 

The Python script has a bug which has now been fixed.

amalhub, 2019-07-19:
> > For some reason, the config files are not compatible with Python.
> 
> The Python script has a bug which has now been fixed.

Thank you for supporting to solve the initial issue. Can I please get some help on how to update the configurations as per my requirements given above?

glddiv, 2019-07-19:
@amalhub 
If you want to modify the image resolution and frame rate, you need to modify a series of sensor registers. This is not an area I am good at. I have already submitted your question to the relevant engineer. If there is any new news, you will get a reply.

glddiv, 2019-07-22:
Hi @amalhub 
This is a 1280x720 configuration file.
[OV5642_1280x720_Config_File.zip](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/files/3415874/OV5642_1280x720_Config_File.zip)

amalhub, 2019-07-23:
Hi @glddiv 
Thank you for the configurations.
Do you have any documentation on how to configure the sensor registry values? 
Are you using a standard IC or some custom made one? If I could get my hands on the datasheet or a manual on setting the registry values, it would be great for my project work.

ArduCAM, 2019-07-23:
You can find how to read/write sensor register from the information below:
https://www.arducam.com/docs/usb-cameras/usb-camera-shield/software-sdk-and-api-for-c-c/ 
OV5642 datasheet can be found below:
www.arducam.com/downloads/modules/OV5642_V1.11.pdf

mitchgalea, 2019-07-24:
Hi @ArduCAM 
The second link does not seem to work. When I access the pdf file I get the 404 response from your website. Could you please recheck and send me the download link?

amalhub, 2019-07-24:
Hi @ArduCAM
The second link does not seem to work. When I access the pdf file I get the 404 response from your website. Could you please recheck and send me the download link?

ArduCAM, 2019-07-24:
It should be https://www.arducam.com/downloads/modules/OV5642/OV5642_V1.11.pdf  
Sorry for the inconvenience.

amalhub, 2019-07-24:
Thanks for all the support! :+1:

closed 2019-07-24 by amalhub