ArduCAM/ArduCAM_USB_Camera_Shield #113: Vertical stripes in Image

lionfish0, 2019-07-20:
Hello, this is a relatively minor issue, but I'm finding the image has vertical stripes in.
[image]
Using the AR0135 monochrome, with a flash, global shutter. I can't remember what the exposure time was (maybe 1ms?)

In particular on the left hand side I notice it more in the image.

I know there are individual column gains that the camera works out - maybe it's something to do with that?

Oddly some of the columns are brighter only every other pixel (like it's just certain colour channels?)... ah, I've realised that I've not set the colour gains equal, https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/issues/84#issuecomment-493704488. I'll try that.

But I don't know if that will resolve the columns appearing...

The pattern is static (doesn't change between images) so I could probably work out what it is and subtract (or divide) it out of the image.
Just wondering if there's a better way, maybe I'm missing something. E.g. maybe the column gain correction thing isn't enabled?

In case it's relevant here's my json config file.
AR0135_1280x964_ext_trigger_M.json
```
{
	"camera_parameter" : {
		"CFG_MODE":1,
		"TYPE":"AR0135",
		"SIZE":[1280,964],
		"BIT_WIDTH":8,
		"FORMAT":[4,0],
		"I2C_MODE":3,
		"I2C_ADDR":"0x20",
		"TRANS_LVL":64
	},
	"board_parameter" : [
	],
	"board_parameter_dev2" : [	
		["0xd7","0x4600","0x0100","0x01",["0x05"]],
		["0xf6","0x0000","0x0000","0x03",["0x03","0x04","0x0c"]]
	],
	"board_parameter_dev3_inf2" : [	
		["0xf3","0x0000","0x0000","0x00",[]],
		["0xf9","0x0000","0x0000","0x00",[]]
	],
	"board_parameter_dev3_inf3" : [	
		["0xf3","0x0000","0x0000","0x00",[]],
		["0xf9","0x0000","0x0000","0x00",[]]
	],
	"register_parameter" : [	
		["0x3028","0x0010"],
		["0x302e","0x0002"],
		["0x3030","0x002f"],
		["0x302c","0x0001"],
		["0x302a","0x0008"],
		["0x3032","0x0000"],
		["0x30b0","0x0480"],
		["0x301a","0x10dc"],
		["0x301a","0x1990"],
		["0x300c","0x06c2"],
		["0x3002","0x0000"],
		["0x3004","0x0000"],
		["0x3006","0x03bf"],
		["0x3008","0x04ff"],
		["0x300a","0x03de"],
		["0x3012","0x0040"],
		["0x3014","0x00c0"],
		["0x30a6","0x0001"],
		["0x308c","0x0000"],
		["0x308a","0x0000"],
		["0x3090","0x03bf"],
		["0x308e","0x04ff"],
		["0x30aa","0x03de"],
		["0x3016","0x0064"],
		["0x3018","0x00c0"],
		["0x30a8","0x0001"],
		["0x3040","0x8000"],
		["0x3064","0x1982"],
		["0x31c6","0x8000"],
		["0x3100","0x0000"],
		["0x305e","0x0020"],
		["0x3046","0x0100"],
		["0x3012","0x0100"]
	],
	"register_parameter_dev3_inf2" : [	
		["0x3030","0x0020"]
	],
	"register_parameter_dev3_inf3" : [	
		["0x3030","0x003a"]
	]
}
```

ArduCAM, 2019-07-22:
Have you tested our Windows GUI and the default config comes with it?

betaBison, 2020-08-25:
I have a similar issue. In the Windows GUI the image from the AR0135 does not have vertical stripes, but when I use the exact same configuration file on a Raspberry Pi, the image does have vertical stripes. Is there a known solution?