ArduCAM/ArduCAM_USB_Camera_Shield #116: After script restart Arducams are not working with ROS implementation

amalhub, 2019-08-02:
Hi, 

I have purchased several Arducam OV5642 cameras (UC-234 Rev.B) with Arducam USB camera shields (UC-391 Rev.D) to connect them into my Ubuntu 16.04 pc. When I connect multiple cameras and run the ROS launch after setting proper configurations; at the first instance, it runs smoothly without any issue. Once I stop running the scripts and restart, it won't publish any images to the ROS topics. 

To overcome this issue I have to disconnect the USB cables and plug it back and then restart the ROS launch. What is the cause of this issue? Is there a solution to avoid unplug and reconnect the USB connections? 

Herewith I have attached my configs as well. Please help me with this issue.
[launch_config.zip](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/files/3459375/launch_config.zip)

glddiv, 2019-08-05:
@amalhub 
ROS has the same bug as python script in question #112 .
It has been fixed and will soon be updated to github.

amalhub, 2019-08-06:
Hi @glddiv 
I took the latest code from github and tested. Now the cameras can be shutdown and restart without any issue. But I noticed a different issue now. 
When I move the cameras individually the video streaming runs smoothly but, **when I start moving both cameras together, one camera stream starts failing and starts skipping frames** with errors in ROS backlog. 
Can you please check on this issue?
Herewith I have attached the ROS backlog error.
[image]

glddiv, 2019-08-06:
@amalhub 
Can you try to replace the dynamic library under ROS with the dynamic library in the Python example? The dynamic library in ROS seems to be missing from the update.
One more thing, please reduce "TRANS_LVL": 128 in the configuration file to "TRANS_LVL": 64

amalhub, 2019-08-07:
Hi @glddiv 
Thank you very much! The suggested changes are working and now I can stream from multiple cameras. 
Could you please let me know what is the **maximum resolution** I can configure to run **3 cameras** simultaneously in **30fps** speed? 
Will you be able to share the configuration file for that maximum resolution setting?

glddiv, 2019-08-09:
@amalhub 
[OV5642_JPEG_1920x1080.zip](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/files/3485318/OV5642_JPEG_1920x1080.zip)

amalhub, 2019-08-13:
Thank you very much for the support given.

closed 2019-08-13 by amalhub