ArduCAM/ArduCAM_USB_Camera_Shield #117: 3 Camera synchronization using external trigger

pritorius, 2019-08-02:
I need some help about on using Arducam.
I have 3 MT9V034 cameras with USB camera shield (UC-391 Rev D). I need to setup these 3 cameras to be synchronized.

As far as I understand, I need to use the following code:
https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/blob/master/Linux_x86/Cpp/External_trigger_demo/ArduCam_Ext_Trigger_Demo.cpp
with config file:https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/blob/master/cpp_config/MT9V034_VGA_Stereo.yml

but I am little confused about how the hardware setup should be.
As far as I understand:
1. All 3 cameras should be set in External Trigger mode.
2. The "External Trigger interface of the UC-391 Rev D of the middle camera should be attached to the trigger input of the left and right camera board ( Please find attached image - the 2 thick purple lines are connected from middle camera usb shields external trigger pin to left and right camera boards trigger input).
3. So, when I call read Image on the middle camera, the external trigger from the middle camera should also fire on the left and right camera.
4. I read images from all 3 camera buffer.

[image]

Will this work?

Razbotics, 2019-08-03:
Did your idea work? Also did you tried triggering camera at 20 to 30 hz? I want to run visual inertial odometry algorithm.

glddiv, 2019-08-07:
> I need some help about on using Arducam.
> I have 3 MT9V034 cameras with USB camera shield (UC-391 Rev D). I need to setup these 3 cameras to be synchronized.
>
> As far as I understand, I need to use [...]

You need to connect the trigger pins of the three "USB camera shields" together and send an external trigger signal to the "USB camera shield" using a microcontroller or something else.
 (Note: The type of trigger signal is a rising edge)
One more thing is that you should use [MT9V034_VGA.yml](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/blob/master/cpp_config/MT9V034_VGA.yml) instead of MT9V034_VGA_Stereo.yml

pritorius, 2019-08-07:
@glddiv 
> You need to connect the trigger pins of the three "USB camera shields" together and send an external trigger signal to the "USB camera shield" using a microcontroller or something else.
> (Note: The [...]

So, I did the following:
1. I connected the the camera boards trigger and ground pin ( NOT the USB shields trigger and ground pin) to the pin 8 and GND of an Arduino as follows:
[image]

2. I am sending rising edge signal for 1 millisecond to the Pin 8 of Arduino every 2 seconds.
3. Then when I start running the External Trigger Demo, I am getting an image every 2 seconds:
[image]

So I guess that's it! 
One thing I am confused about is you said to use the USB Camera shield's TRIGGER and GND pin. But I could not make that work. I had to use the camera board's TRIGGER and GND.

Now I need to try next:
1. Connect all 3 cameras 2 the PIN 8 and GND of Arduino.
2. Try to get image from all 3 cameras at 30 HZ.

Razbotics, 2019-08-07:
Aren't those trigger pins on camera as well as usb camera shield has a common line?

glddiv, 2019-08-08:
> One thing I am confused about is you said to use the USB Camera shield's TRIGGER and GND pin. But I could not make that work. I had to use the camera board's TRIGGER and GN

Sorry, my description is not strict enough. The different versions of the "USB Camera shield" TRIGGER PIN are not the same. The newer version of the "USB Camera shield" is configurable. We are working on the documentation for the external trigger.

f5-zahmed, 2019-08-13:
> Sorry, my description is not strict enough. The different versions of the "USB Camera shield" TRIGGER PIN are not the same. The newer version of the "USB Camera shield" is configurable. We are [...]

Thanks for the reply.
Is it possible to update the library ( or get a version of the library binary) in such a way that if I send a soft trigger to one usb camera shield is sent, the trigger pin of the usb board is also triggerd?
[image]

So, From the above picture, if I send a soft trigger from my code ( the black arrow), the picture on the attached camera will be taken and also, the external trigger pin ( by red arrow) will go up so that I can wire that pin to 2 other cameras and take picture simultaneously. In that way I will not need to maintain an arduino.

please let me know if this is possible.

glddiv, 2019-08-21:
@f5-zahmed 
Unfortunately, this feature cannot be implemented by simply modifying the SDK. Hardware support is required to implement this feature.

f5-zahmed, 2019-08-21:
@glddiv 
> t be implemented by simply modifying the SDK. Hardware support is
Thanks for the reply. I will close this issue.