ArduCAM/ArduCAM_USB_Camera_Shield #119: Error capture image, rtn_val = 65318 for OV7251

chris24sahadeo, 2019-08-12:
Hey All, we are getting this issue when using `OV7251_MIPI_RAW10.yml` config for the following setup:

- [Arducam OV7251 MIPI 0.31MP Monochrome Global Shutter Camera Module for Raspberry Pi](https://www.arducam.com/product/arducam-ov7251-mipi-0-31mp-monochrome-global-shutter-camera-module-raspberry-pi/)

- [ArduCAM USB3.0 Camera Shield
](https://www.arducam.com/product/usb3-0-camera-shield/)

- Ubuntu 16.04.

- We are trying to run the StreamingDemo example.

Command:
`./ArduCam_Demo ../../../cpp_config/OV7251_MIPI_RAW10.yml`

Output:
```device num:1
index:   0	Serial:AU3S-1901-0005
Serial: AU3S-1901-0005
capture thread create successfully.
read thread create successfully.
Capture began, rtn_val = 0
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
Error capture image, rtn_val = 65318
```

Any help would be appreciated

glddiv, 2019-08-19:
Hi @chris24sahadeo 
Sorry to reply to you so late, we are replacing the linux configuration file format with .cfg, which should solve your problem.

Please try this example: [Streaming_demo.zip](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/files/3513998/Streaming_demo.zip)
(command: `sudo ./ArduCam_Demo ./OV7251_MIPI_RAW10.cfg`)

In addition to the dependencies before Streaming_demo, you also need to install [arducam_config_parser](https://github.com/ArduCAM/arducam_config_parser).

chris24sahadeo, 2019-08-20:
Hey @glddiv  

The new `Streaming_demo.zip` that you included works!   

We had to delete the included executable and run `make` to generate a new executable and that got rid of the `Error capture image, rtn_val = 65318` error. The `Streaming_demo` works fine now.  

We would like to use the camera with ROS. Do you have an updated ROS package that uses the new `.cfg` file? If yes then kindly send it our way.

Much thanks for the assist.

glddiv, 2019-08-21:
@chris24sahadeo 
I have not fully tested it, please let me know if there are any problems.
[arducam_usb2_ros.zip](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/files/3523152/arducam_usb2_ros.zip)

chris24sahadeo, 2019-08-22:
@glddiv **The ROS package you sent works :)** 

_Minor note:_ 
Initially when running the  `arducam_node.launch` file and `rosservice call /cam0/arducam/capture`, we ran into the following exception from CV Bridge: 
```
encoding specified as bgr8, but image has incompatible type 8UC1
```
Changing `"bgr8"` to `"8UC1"` in line 194 of the `arducam_ros_node.py` file fixed this error.

---

Subscribing to the `/cam0/arducam/camera/image_raw` topic in RViz works well even without the aforementioned tweak.

---

The only remaining concern is that the image from the camera appears dark (underexposed).
Not sure if this is a hardware limitation of the camera itself, or if it can be tweaked in the `.cfg` file. 
If you can shed some light on this subject, it would be much appreciated.

Otherwise, we are up and running!

Thanks again for all the help.

glddiv, 2019-08-23:
@chris24sahadeo 
Yes, the exposure time can be adjusted by modifying some registers (adjusted in the configuration file or adjusted in the program). Please check the sensor data sheet for which registers need to be adjusted.
[OV7251 Datasheet](https://datasheet4u.com/datasheet-pdf-file/1252892/OmniVision/OV7251/1)
This is a description of the OV7251 exposure register:
[image]